google/xls #115: Fuzz bug: incorrect "one_hot msb" transform

cdleary, 2020-08-24:
Found by fuzz test while tweaking something unrelated, and I minimized it a bit by hand since the current minimizer only works at the IR level. First time we've seen the fuzzer catch something in a while! As a mitigation for now I'll disable the transform until we properly understand the bug and can re-enable the pass.

It also would be nice if we could separate out the transform from the BDD analysis somehow, right now it's monolithic so we have to internally disable that part within the mega BDD transform pass.

```
// options: {"input_is_dslx": true, "convert_to_ir": true, "optimize_ir": true, "codegen": false, "simulate": false, "simulator": null}
// args: bits[8]:0x5; bits[12]:0xe9c; bits[35]:0x5_5540_0000

fn main(p3: u8, p6: u12, p8: s35) -> u8 {
  let x13: u12 = clz(p6);
  let x14: u12 = clz(x13);
  let x18: u8 = ((p8 as u8)) | (p3);
  let x24: u1 = xor_reduce(x14);
  let x30: u8 = one_hot_sel(x24, [x18]);
  x30
}
```

Assigning to Mark since he's most familiar with the pass.

jbaileyhandle, 2020-08-25:
I'll try to tackle this while I figure out the github dev flow.

cdleary, 2020-08-25:
Awesome! Thanks Jonathan. Some docs in https://google.github.io/xls/#building-from-source but let me know if they can use augmenting! I think for pull requests you just click "make a fork" and then "make a pull request" when your branch is ready, and we ask that people try to squash their branches into single commits before making pull requests because it makes the history cleaner: https://google.github.io/xls/contributing/#pull-request-style

PR #121 (2020-08-29): Fixed bug in BDD simplification pass. 
The bug occured because we reused BDD analysis generated before OneHot MSB simplification after OneHot MSB simplification. Although the OneHot transform is only performed if the output is unaffected, this transform may change intermediate values. The previous BDD analysis on such intermediate values may therefore no longer be accurate. Fixed by performing a fresh analysis after the OneHot Transfrom.

Fixes #115
closed 2020-08-29 by copybara-service[bot]