google/xls #116: Hoist JIT to xls/jit/

cdleary, 2020-08-25:
Right now JIT things are mixed in the IR directory, but as discussed feels more like a top-level project component (an aspect that consumes the central IR, akin to solvers or codegen) and the number of JIT things is growing over time. Could make sense to factor out an `xls/interp/` too by that reasoning, but don't mind either way since the `Value` type needs to be present in the IR for constant (literal) values anyway.

Also the organization could have some knock-on benefits -- you could conceive of running tests without JIT for faster OSS presubmits, and you can `blaze query 'deps(xls/dslx/..., xls/jit...)'` with the JIT things split out into their own location to query for interpreter-only tests.

RobSpringer, 2020-08-25:
On it!

RobSpringer, 2020-08-28:
Done!

closed 2020-08-28 by RobSpringer