google/xls #120: DSLX-test JIT does not JIT the test fn bodies, only invoked functions

cdleary, 2020-08-28:
Closing this out, though going to create a derived issue for the (smaller task) remaining bits:

1. We don't translate the test function body itself (the one generally containing `assert_*` builtin calls), only function calls (via `Invocation` nodes) that are reached from the test function entrypoint.
2. We don't reuse JIT objects via our Python bindings; e.g. would be useful to amortize overhead in QuickCheck, we have to re-JIT every time at the moment I believe.

_Originally posted by @cdleary in https://github.com/google/xls/issues/10#issuecomment-682260404_

hmontero1205, 2020-08-28:
Some notes for future reference:

- The hard part about translating the test function body itself is that it is likely to be filled with `assert_*`s and other DSLX-only functions. We would also want to translate the `assert_*`s arguments.

- In order to reuse LlvmIrJit C++ objects, we must first wrap the LlvmIrJit C++ object type in PyBind. This way, we can break `CreateAndRun(fn, args)` into a static method `Create(fn)` and an instance method `Run(args)`.

- I envisioned a cache in the interpreter for JIT objects, mapping the ir mangled function name to the the JIT object.

allight, 2026-11-14:
Closing this issue as fixed and made obsolete by changes in how we run tests and generate JITs (accepted by @allight): `IrJitTestRunner` (`xls/dslx/run_routines/ir_test_runner.cc`, selected via `--evaluator=ir-jit` in `xls/dslx/interpreter_main.cc:167-168`) converts and JIT-compiles DSLX test functions and test procs directly, and `RunComparator` (`xls/dslx/run_routines/run_comparator.h:72-82`) caches compiled `FunctionJit` instances in `jit_cache_`. (Closed by Gemini)

closed 2026-11-14 by allight