google/xls #123: [App] Implement more floating point operations in modules/

cdleary, 2020-08-28:
We've been building up verified IEEE FP32 operations (and also some bf16 operations); e.g.

* fpadd, fpmul in https://github.com/google/xls/tree/6e21d521d32fd484690664d81309631f07edab55/xls/modules
* bf16 "next float" style significand bumping: https://github.com/google/xls/blob/1b6859dc384fe8fa39fb901af3de8453661ff345/xls/dslx/stdlib/bfloat16.x#L29

It would be interesting to comb through the IEEE FP32 specification and implement some more of the primitive operations.

/cc @majnemer if there are any particularly interesting ones you might recommend

meheff, 2023-03-06:
FYI @hzeller 

One option is to implement some rounding functionality (ceil, floor,etc). Possbilities from cmath: https://en.cppreference.com/w/cpp/header/cmath

(context: to familiarize himself with DSLX Henner was interested in implementing something useful but not too big)

cdleary, 2023-04-12:
One other aspect of this project is that if we implement more well verified parametric FP modules we'll probably want to make sure there's a path to linking them with xlscc translation units so we don't need implementations in both universes for these more complex and subtle library-like entities. cc @grebe who had been thinking about our generalized linkage e.g. in #864

proppy, 2026-10-28:
I believe https://github.com/google/xls/blob/main/xls/dslx/stdlib/apfloat.x leveled up quite a bit since then, but I think @cdleary originally statement still holds:

> It would be interesting to comb through the IEEE FP32 specification and implement some more of the primitive operations.