google/xls #132: build failure for bazle py_library check

ckmufeng, 2020-09-03:
when i try to run  `bazel run -c opt //xls/dslx/interpreter:interpreter_main -- /tmp/simple_add.x`.
I get error below:

ERROR: /sata/milo.ck/.cache/bazel/_bazel_milo.ck/c2f3bec9ac81f15fc6283f7e3cc06fc7/external/xls_pip_deps/requirements.bzl:19:9: Traceback (most recent call last):
        File "/sata/milo.ck/0_Projects/xls/xls/dslx/interpreter/BUILD", line 137
                py_library(name = "interpreter", <3 more arguments>)
        File "/sata/milo.ck/0_Projects/xls/xls/dslx/interpreter/BUILD", line 147, in py_library
                requirement("termcolor")
        File "/sata/milo.ck/.cache/bazel/_bazel_milo.ck/c2f3bec9ac81f15fc6283f7e3cc06fc7/external/xls_pip_deps/requirements.bzl", line 19, in requirement
                fail(<1 more arguments>)
Could not find pip-provided dependency: 'termcolor'

After I comment all requirement in py_library, it seems right.

cdleary, 2020-09-03:
Thanks for reporting! Is this the docker container environment (`docker build .` at the root of the repo) or a different environment?

ckmufeng, 2020-09-04:
Yes. It's docker container environment (`docker build .`)

ckmufeng, 2020-09-04:
Maybe i use wrong docker environment!

closed 2020-09-04 by ckmufeng