google/xls #133: Seg-fault for large chain of xors

renau, 2020-09-03:
The following python program creates a chain of xors. It works for 5K xor chain, but seg-faults for 10K.

```

def main():
    import subprocess
    nums = [5000, 10000]
    for num in nums:
      f = open("xor_" + str(num) + ".x", "w+")

      for i in range(num):
          if num == 1:
              f.write("fn xor(x: u2, y: u2) -> u2 {\n")
              f.write("    x ^ y\n")
              break
          if i == 0:
              f.write("fn xor(x: u2, y: u2) -> u2 {\n")
              f.write("    let t0 = x ^ y;\n")
          elif i == num-1:
              f.write("    t%d ^ x\n" % (i-1))
          else:
              f.write("    let t%d = t%d ^ x;\n" % (i, i-1))

      f.write("}")
      f.close()
      # bashCommand =  "prp xor_" + str(num) + ".prp | sort -n -tK -k2 > y && column -t y > x && rm -f y && mv x xor_" + str(num) + ".cfg"
      # subprocess.check_call(bashCommand, shell=True)
if __name__ == "__main__":
    main()
```

cdleary, 2020-09-03:
Thanks for filing! I'll try to repro, but can you provide the traceback / dump info as well?

renau, 2020-09-03:
I closed the docker image, but the text (still in the terminal was)
```
...
  File "/home/user/projs/synth/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/bindings.py", line 76 in resolve_node                 
File "/home/user/projs/synth/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/bindings.py", line 76 in resolve_node                 
File "/home/user/projs/synth/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/bindings.py", line 76 in resolve_node                 
File "/home/user/projs/synth/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/bindings.py", line 76 in resolve_node                 
File "/home/user/projs/synth/xls/bazel-bin/xls/dslx/ir_converter_main.runfiles/com_google_xls/xls/dslx/bindings.py", line 76 in resolve_node                 
Segmentation fault     
...                                                                                                                                                        
```

The command that I executed was:
```
./bazel-bin/xls/dslx/ir_converter_main ./xor_10000.x >pp.ir
```

cdleary, 2020-09-03:
Thanks! Expect I can try to repro this afternoon.

cdleary, 2020-09-03:
Looks like this is a CPython stack overflow because of high recursion depth, it's overflowing the native thread stack with interpreter call frames.

I'm actively working on switching the DSL parser to C++ (and in addition, working on avoiding let bindings being recursive expressions), but in the meantime depending on the nature of your microbenchmark you might consider targeting the IR layer directly, like so:

```
def main():
    import subprocess
    nums = [5000, 10000]
    for num in nums:
      f = open("xor_" + str(num) + ".ir", "w+")

      for i in range(num):
          if i == 0:
              f.write('package xor_ubench\n')
              f.write("fn xor(x: bits[2], y: bits[2]) -> bits[2] {\n")
              f.write("    t0: bits[2] = xor(x, y)\n")
          else:
              f.write("    t%d: bits[2] = xor(t%d, x)\n" % (i, i-1))

      f.write("}")
      f.close()
      
if __name__ == "__main__":
    main()
```

```
$ ./bazel-bin/xls/tools/parse_ir ~/xors/xor_10000.ir
$ time ./bazel-bin/xls/tools/opt_main ~/xors/xor_10000.ir 
package xor_ubench

fn xor(x: bits[2], y: bits[2]) -> bits[2] {
  ret param.2: bits[2] = param(name=y)
}

real    0m16.769s
user    0m16.473s
sys     0m0.290s
```

I'd also note we haven't done much to optimize compile times for microbenchmarks, but would in any case be happy to see any results you're looking at!

hundt98847, 2020-09-03:
In the meantime - if this serves a purpose other than benchmarking - we could grow DSLX's stack. Something like:

```
import resource
import sys

# print resource.getrlimit(resource.RLIMIT_STACK)
# print sys.getrecursionlimit()
# print

resource.setrlimit(resource.RLIMIT_STACK, [0x10000000, resource.RLIM_INFINITY])
sys.setrecursionlimit(0x100000)

```

cdleary, 2020-09-03:
Nice -- we already do setrecursionlimit (and I tried upping that even more) but it didn't help because it was the native stack, I didn't try changing the rlimit (I don't think I knew you could from within Python! TMYK). Will try that and see if it helps.

cdleary, 2020-09-03:
Looks like that worked for me. I just put it at the start of `ir_converter_main.py`. Awesome ref @hundt98847!

That being said, doing all that parsing and IR conversion in CPython is a leeeetle slow...

```
$ time ./bazel-bin/xls/tools/parse_ir ~/xors/xor_10000.ir 
real    0m0.365s
user    0m0.343s
sys     0m0.018s
$ time ./bazel-bin/xls/dslx/ir_converter_main ~/xors/xor_10000.x
real    2m39.663s
user    2m39.135s
sys     0m0.363s
```

Since we use deeply recursive things today, but we expect to move to a C++ based parser soon, I'll put a generous stack rlimit bump into our standard prologue for `py_binary`s.

renau, 2020-09-03:
I think that it is solved until the C++ parser. I use a small set of synthetic tests to check scalability (LiveHD, FIRRTL, yosys...), so no practical reason to xor a chain beside benchmarking and testing the tool.

closed 2020-09-03 by renau