google/xls #135: Consider using .gitattributes to get GitHub to highlight .x files as e.g. Rust.

eddyb, 2020-09-08:
Right now they're auto-detect as Logos, it seems like.
According to https://github.com/github/linguist#using-gitattributes, you could resolve this with a `.gitattributes` file containing:
```gitattributes
*.x linguist-language=Rust
```
That's assuming Rust is the closest supported language, syntax-wise.
For example, [this is what `riscv_simple.x` looks like](https://gist.github.com/eddyb/1896fe441ab385e26913fdb385bc5175) with Rust syntax highlighting.

cdleary, 2020-09-09:
Awesome suggestion, thanks so much! Will see if I can add this tomorrow.

cdleary, 2020-09-09:
Merged, thanks much!

closed 2020-09-09 by cdleary

eddyb, 2020-09-09:
I was worried it wasn't kicking in so I kept checking, but it's working now! Examples: https://github.com/google/xls/blob/d286be0c24f33eb265fa9e44895e1542da233e82/xls/examples/riscv_simple.x#L121-L123 https://github.com/google/xls/blob/d286be0c24f33eb265fa9e44895e1542da233e82/xls/examples/crc32.x#L29-L31
The second one, notably, hasn't been changed in a month, so it does apply retroactively.

cdleary, 2020-09-11:
Yeah for github pages as well I often see what seems like very aggressive caching behavior.

Looking sooo much nicer now, thanks again for filing the suggestion/issue!