google/xls #139: compilation error with bazel

HolyLow, 2020-09-14:
My bazel version is 3.5.0, and when I follow the guide in README: bazel test -c opt ... , error occurs:
```
ERROR: Skipping '../..': Bad target pattern '../..': package name component contains only '.' characters
WARNING: Target pattern parsing failed.
ERROR: Bad target pattern '../..': package name component contains only '.' characters
INFO: Elapsed time: 0.126s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded)
FAILED: Build did NOT complete successfully (0 packages loaded)
```
It seems to be some character error, but I am not familiar with bazel... Any suggestions?

cdleary, 2020-09-14:
Hi, is it possible you're trying to do a relative-path style build?

In Bazel everything is rooted in `//`, so you can do `bazel test -c opt //...` from any subdirectory under the repo root and it should build and test everything.

Can you provide more details on the commands you're running? You can also check out the [docker container via the Dockerfile](https://google.github.io/xls/#building-from-source) which provides instructions that should work on any machine you can install docker onto. Thanks!

HolyLow, 2020-09-14:
The  command I use is:
```
$ git clone https://github.com/google/xls.git
$ cd xls
$ bazel test -c opt ...
```
Should I move the project to somewhere else? Currently the project locates in $HOME/project/xls . I don't know the path of // of bazel as well...

cdleary, 2020-09-14:
Thanks for explaining.

The `//` I mentioned means "bazel root" -- bazel interprets that as "relative to the directory where the `WORKSPACE` file is located", so you can be in any directory underneath where that `WORKSPACE` file is and run the same command (it's like an absolute path relative to the directory root). `...` is the "all targets recursively in any subdirectory" wildcard, so when we do `bazel test -c opt //...` we're saying "do optimized-build testing of all target under the repo root". So you should be able to run that in any directory within the repo, the commands you listed look like they should work properly for finding targets.

**A few troubleshooting ideas:**

* What is your operating system (`cat /etc/issue`) and output of `bazel --version` (you mentioned 3.5 just double checking)?
* What is the output of `echo $SHELL`
* Try `bazel build -c opt //...` instead?
* Check that `ls WORKSPACE` shows you're currently in the Bazel root directory
* Try `bazel query ...` and `bazel query xls/tools:all`
* Try building a single target; e.g. `bazel build -c opt //xls/ir:ir`
* Try testing a single target; e.g. `bazel test -c opt //xls/ir:bits_test`

If none of that yields useful information we can keep on trying to debug, but do you think it is an option to try the docker-based build? (The benefit is it's hermetic and reproducible so confounding environmental factors are less of an issue.)

HolyLow, 2020-09-14:
Thanks for your advices, and it worked! 
My os is Ubuntu 20.04 launched by WSL2, win10. My shell is zsh. I tried ```bazel build -c opt //...``` and it worked!

closed 2020-09-14 by HolyLow

cdleary, 2020-09-14:
Glad to hear, though TBH I'm not sure what was happening before -- I don't think `zsh` does an expansion for `...` (I tried it out on my local machine, seemed to work ok). Perhaps there's some custom expander / macro in your environment? Not sure, but in any case, glad it's working now! Thanks for letting us know.