google/xls #143: Create standard Python unit testing base class for Python-based testbenches

cdleary, 2020-09-22:
Often people want to bail out to flexible non-synthesizable testbenches, and there's some boilerplate scaffolding currently required to do that, loading up the IR files and executing them and such. We should be able to hide it all inside of a base class, similar to how we did in the C++ test bench / JIT header generation scaffolding.

cdleary, 2021-02-17:
Since we have our `test_base` library in xls/common that calls `init_xls` appropriately, the adjusted task description here is probably to switch all of our Python tests to use that and make a "METADATA-file" (Goog internal presubmit check description) based warning for use of the raw underlying libraries.

cdleary, 2021-02-17:
Good use: https://github.com/google/xls/blob/62deead76f19000e143dd3dae47f13c84502815a/xls/ir/python/type_test.py#L24
Bad use: https://github.com/google/xls/blob/c41a2732b3a6918edd497fd118156d1214e5fb57/xls/ir/python/number_parser_test.py#L23

dplassgit, 2025-05-13:
These tests were removed.

closed 2025-05-13 by dplassgit