google/xls #146: IR visualizer mode that shows scheduled pipeline stages

cdleary, 2020-09-24:
Right now the visualizer shows critical path /before/ scheduling and register minimization.

Sometimes you tweak the frequency and you're surprised at how much can fit into fewer cycles as a result! Having a way to view the pipe stages explicitly (post sched) and which nodes have been offset from their ASAP-scheduling time (according to the critical path) on account of discrete pipeline stages (i.e. not fitting into the prior stage, so having to take the delay up to flop time) would help to make diffs of critical paths for reasonable number of pipe stages more clearly visible.

Speculatively assigning to Mark since I know he wanted to add this, wanted to make an issue we could reference going forward.

meheff, 2022-01-31:
cc @nanavati

tcal-x, 2023-07-14:
I think there has been some progress on this: the arrow between two nodes is dashed if it crosses a pipeline stage boundary.   However, to me, it is still difficult to see the distinct stages, since the nodes don't get correctly aligned so that all of the nodes in one stage are above all of the nodes of the following stage.