google/xls #149: Optimize deconstruct->reconstruct tuple patterns

meheff, 2020-09-28:
Example:

t = ...
tuple_index.0 = tuple_index(t, 0)
tuple_index.1 = tuple_index(t, 1)
...
tuple_index.n = tuple_index(t, n)
tuple.x = tuple(tuple_index.0, ... , tuple_index.n)

tuple.x can just be replaced with t.

rajatdiptabiswas, 2020-10-07:
Hi @meheff 
Would you kindly let me know what exactly needs to be done so that I can try working on it?

meheff, 2020-10-09:
@rajatdiptabiswas Thanks for your interest! What needs to be done is the pass [tuple simplification](https://github.com/google/xls/blob/main/xls/passes/tuple_simplification_pass.cc) needs to be extended to match the pattern shown in the top comment.

Basically you need to iterate through each Tuple operation in the function (nodes with opcode Op::kTuple) and check to see if it matches a pattern:
- each of its operand is a TupleIndex operation
- the index of each TupleIndex operand matches the operand number
- the operand of all the TupleIndex operands is the same.
It will look like

```
 foo = Tuple({
     TupleIndex(bar, 0),
     TupleIndex(bar, 1),
     TupleIndex(bar, 2),
     ... })
```

Then you can replace foo with bar because they are effectively the same value.

The compiler XLA which is uses a similar dataflow-based IR has a similar optimization which can be seen here:

https://github.com/tensorflow/tensorflow/blob/master/tensorflow/compiler/xla/service/tuple_simplifier.cc#L36

Let me know if you have any questions.

rajatdiptabiswas, 2020-10-11:
Sorry, I didn't see this earlier. 
I shall try my best to get it done. Thanks for the explanation!

rajatdiptabiswas, 2020-10-12:
Hello @meheff 

I went over the code links you sent but I'm having trouble understanding most of it.

What are the prerequisites I should know before attempting to fix this?
Should I set the development environment using `bazel` or Docker first?

As far as I can understand, I have to edit `tuple_simplification_pass.cc` so that the case you mentioned is also checked for in the file. I came across the `abseil` library as well. Will going through the `abseil` docs help?

If you could point out any references or docs I could look into, it would be really helpful.

Either way, I think it's better someone more experienced looks into the issue. 
They'll definitely be able to get it done much faster.

Thanks in advance!

cdleary, 2020-10-19:
Hi @rajatdiptabiswas, no worries at all! Is there an area you're most interested in learning about so we might be able to come up with a better first bug for you to check out? Were you able to follow the readme instructions to get the docker image working and bazel test showed all the tests passing? Feel free to reach out on the xls-dev mailing list linked from the README for more discussion out of band of this bug if you'd like!

felixzhuologist, 2020-10-24:
I have a version of this at: https://github.com/google/xls/compare/main...felixzhuologist:tuple but i'm struggling to find cases where adding this handles a case that the pre-existing optimization doesn't.

for example, in:
```
 fn func(x: bits[1], y: bits[2], z: bits[3]) -> (bits[1], bits[2], bits[3]) {
  tuple.1: (bits[1], bits[2], bits[3]) = tuple(x, y, z)
  tuple_index.2: bits[1] = tuple_index(tuple.1, index=0)
  tuple_index.3: bits[2] = tuple_index(tuple.1, index=1)
  tuple_index.4: bits[3] = tuple_index(tuple.1, index=2)
  ret tuple.5: (bits[1], bits[2], bits[3]) = tuple(tuple_index.2, tuple_index.3, tuple_index.4)
 }
```
the current code already optimizes this to 4 nodes (the 3 params and the return): i think the current tuple simplification replaces the return with `tuple(x, y, z)`, and then DCE removes `tuple.1`.

The only case I can think of that doesn't currently get optimized, is if the tuple that gets reconstructed is passed directly as a parameter like : `fn func(data: (bits[1], bits[2], bits[3])) -> (bits[1], bits[2], bits[3])` (i.e. the elements of the tuple don't exist as their own values in the scope). Are there other examples where this optimization is useful?

cdleary, 2020-10-24:
Hi @felixzhuologist -- I think we need to start taking locks on the issues by taking the assignment field since I think Vincent also had a change in flight to do this (whoops!).

That last paragraph makes sense that it would be one of the cases that make it through that could then be simplified to be a straightforward identity function, I think this optimization is primarily interesting for graph cleanup purposes vs exposing a particular subsequent optimization opportunity.

felixzhuologist, 2020-10-26:
Sounds good, will do that in the future :) I'll leave this one to Vincent then!

felixzhuologist, 2020-11-26:
Marking as closed since 1571946b243da933d98ec951a349757e0e850c24 seems to address this. Feel free to reopen if there's still work to be done on this

closed 2020-11-26 by felixzhuologist