google/xls #150: ir_parser build error

KOVI89alipes, 2020-10-01:
Hi there, 

I made a clone of the project, build it, but some of the examples are failing. The error is in the ir_parser package

Any thoughts, suggestions?

```
$ /usr/bin/gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.

$ bazel build ir_parser
INFO: Analyzed target //xls/ir:ir_parser (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
ERROR: /home/konstantin.vinogrado/opt/xls/xls/ir/BUILD:288:11: C++ compilation of rule '//xls/ir:ir_parser' failed (Exit 1) gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++17' -MD -MF ... (remaining 52 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
xls/ir/ir_parser.cc: In static member function 'static absl::StatusOr<std::unique_ptr<xls::Package> > xls::Parser::ParsePackage(absl::string_view, std::optional<std::basic_string_view<char> >)':
xls/ir/ir_parser.cc:1252:10: error: could not convert 'package' from 'std::unique_ptr<xls::Package>' to 'absl::StatusOr<std::unique_ptr<xls::Package> >'
   return package;
          ^~~~~~~
xls/ir/ir_parser.cc: In static member function 'static absl::StatusOr<std::unique_ptr<xls::Package> > xls::Parser::ParsePackageWithEntry(absl::string_view, absl::string_view, std::optional<std::basic_string_view<char> >)':
xls/ir/ir_parser.cc:1262:10: error: could not convert 'package' from 'std::unique_ptr<xls::Package>' to 'absl::StatusOr<std::unique_ptr<xls::Package> >'
   return package;
          ^~~~~~~
In file included from xls/ir/ir_parser.cc:15:0:
./xls/ir/ir_parser.h: In instantiation of 'static absl::StatusOr<std::unique_ptr<_Tp> > xls::Parser::ParseDerivedPackageNoVerify(absl::string_view, std::optional<std::basic_string_view<char> >, std::optional<std::basic_string_view<char> >) [with PackageT = xls::Package; absl::string_view = std::basic_string_view<char>]':
xls/ir/ir_parser.cc:1269:76:   required from here
./xls/ir/ir_parser.h:264:10: error: could not convert 'package' from 'std::unique_ptr<xls::Package>' to 'absl::StatusOr<std::unique_ptr<xls::Package> >'
   return package;
          ^~~~~~~
Target //xls/ir:ir_parser failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 2.343s, Critical Path: 2.05s
INFO: 0 processes.
FAILED: Build did NOT complete successfully
```

cdleary, 2020-10-01:
Hi, thanks for noting the GCC version. We generally use C++17 features -- looks like many of them are supposed to be present in 7.5 according to https://gcc.gnu.org/projects/cxx-status.html#cxx17 but this makes it look like maybe some feature isn't there. (The return is moving into an implicit constructor.)

Is it possible to try a newer GCC?

We also have the Dockerfile provided so it's easier to get going without environment issues, see the bottom of https://google.github.io/xls/#building-from-source

HTH!

cdleary, 2020-10-10:
Hi KOVI89alipes, let me know if we can be of further help, I'm thinking of closing this as "not a supported build environment" in a few days. Thanks!

closed 2021-01-08 by cdleary