google/xls #156: EvaluateNodeWithLiteralOperands() should return failed status with non-literal args

spurserh, 2020-10-07:
Right now it asserts / SIGABRTs, which isn't a very clear indication to the programmer using XLS[cc] what they did wrong in their unrolled loop condition.

cdleary, 2020-10-08:
Can you provide the stack trace? Ideally an IR file and inputs would be best.

Did you verify the package/function before you attempted to execute it? The compiler is generally allowed to assume input programs have been verified, since it should be verified (and thereafter rejected if not correctly constructed) in all binaries we provide before it gets to that point.

cdleary, 2020-10-09:
Passing assignment to Sean for sample input / backtrace.

cdleary, 2020-11-14:
Sean said he thinks `add(add(literal, literal), literal)` is enough to trigger the issue.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `EvaluateNodeWithLiteralOperands()` was removed and replaced by `InterpretNode()` (`xls/interpreter/ir_interpreter.cc:70`) and `Translator::EvaluateNode()` (`xls/contrib/xlscc/translator.cc:5984-6002`), which returns `absl::StatusOr<xls::Value>` (`absl::UnavailableError` when `do_check=false` in `xls/contrib/xlscc/translate_loops.cc:755`) instead of asserting or aborting on non-literal loop conditions. (Closed by Gemini)

closed 2026-11-14 by allight