google/xls #157: Wrong optimization (select branch domination?)

spurserh, 2020-10-07:
This IR: 

fn my_package(a: bits[32], b: bits[32]) -> bits[32] {
  literal.3: bits[32] = literal(value=0, pos=1,3,2)
  ne.4: bits[1] = ne(b, literal.3, pos=1,3,2)
  literal.6: bits[32] = literal(value=0, pos=1,8,2)
  sel.5: bits[32] = sel(ne.4, cases=[b, a])
  sel.7: bits[32] = sel(ne.4, cases=[literal.6, sel.5])
  ret identity.8: bits[32] = identity(sel.7, pos=1,2,1)
}

Comes out of opt as this:

fn my_package(a: bits[32], b: bits[32]) -> bits[32] {
  literal.3: bits[32] = literal(value=0, pos=1,3,2)
  ne.4: bits[1] = ne(b, literal.3, pos=1,3,2)
  ret sel.5: bits[32] = sel(ne.4, cases=[b, a])
}

This doesn't make sense to me, as the return is a select between 0 and another value, based on b != 0. How does that become a select between only a and b, with no possibility of 0?

cdleary, 2020-10-07:
Just an aside as we check this out, you may want to add the "prove unopt equals opt" test target to your flow like we do for the DSLX targets, can be used as a proof confirmation that these are actually the same or that there is a bug. https://github.com/google/xls/blob/55bbbfdf61e4aa6fbe769d7bd2ebf469f8d19aa9/xls/build/build_defs.bzl#L243

meheff, 2020-10-07:
I think the two IR samples are equivalent. The code is equivalent to:

if (b == 0) {
  return 0;
} else {
  return a;
}

The original sample expresses this as:

(b != 0)  ? ((b != 0) ? a : b) : 0

we can assume b != 0 on the left branch of the ternary because it is only selected when b != 0, so we have:

(b != 0) ? a : 0

or:

(b == 0) ? 0 : a

The optimized form is

(b != 0) ? a : b

here we can assume b==0 on the right branch of the ternary because it is only selected when b==0, so we have:

(b ==0) ? a : 0

which is the same as above.

spurserh, 2020-10-07:
The last line is: 

ret sel.5: bits[32] = sel(ne.4, cases=[b, a])

That's (b==0)?a:b, is it not?

meheff, 2020-10-07:
The select case order is opposite of the C++ ternary operator confusinly. I.e.., for a select instruction with a single bit selector, the false case is first and the true case is second. The reason for this is that the select actually treats the selector value as an index (it can be an arbitrary number of bits) into the cases array. This puts the false case (selector == 0) first.

spurserh, 2020-10-07:
The order of the cases is irrelevant, as neither of them are 0?

spurserh, 2020-10-07:
Ohh I see, it's using b as 0. Thanks for taking a look.

closed 2020-10-07 by spurserh

meheff, 2020-10-07:
np!

cdleary, 2020-10-08:
We could specialize that IR formatting by saying on_false=b on_true=a (for that binary case) in case it'd make people less confused.

spurserh, 2020-10-08:
That would be nice: it has been a source of confusion for me the last few days, especially since the FunctionBuilder::Select() ordering is the opposite of in the IR.