google/xls #161: Insert clock gated flops for pipeline stages inserted for non-taken select/switch branches

hongted, 2020-10-15:
In order to reduce power on non-taken select branches, flops inserted for the non-taken branches can be clock-gated.  The idea is that if the select condition is early enough, then additional logic can be put in to gate off the logic.  Idea is summarized in the image:

[image]

XLS can perform analysis during pipeline insertion to infer the enable pin along with the conditions.  Idea of implementation is the following:
   * The one_hot_sel operation establishes a relationship between the select bit and a data wire.
   * Determine the set of nodes which if the data-wire is not selected, do not impact the output (not observable).  One non-exhaustive algorithm is below
      * Determine the set of nodes which are on the fan-in of the data wire.
      * Determine the subset of those nodes in which the data-wire in which any path through that node also goes through the data-wire.
   * Any pipeline stage that is inserted after such a node and before the select, can be gated by the select signal.
   * Additional consideration should be given to the select signal so that it is available early enough not impact timing, and so that the signal is available in the same pipeline stage.Related Work - Observability Don't Care

cdleary, 2022-11-02:
Marking as "long term enhancement" because I believe from the description this is for "fine-grained operand gating" vs the pipeline-based clock gating that we do have support for right now. Correct me if that's not right @hongted

hongted, 2022-11-02:
Somewhat.  Agreed that it is long-term enhancement, but this is more "fine-grained pipeline clock gating" or data-driven pipeline clock gating in that our current pipeline clock gating is based only on whether or not the stage is valid, not whether or not the data is used downstream.

vincent-mirian-google, 2022-11-07:
Just a side note :there are some implications in hardware, for example stability. Clock gating can introduce skew which may not be beneficial in some design.

Perhaps, this can be a user controlled feature at codegen?