google/xls #163: Add range analysis

meheff, 2020-10-19:
We need an analysis which tracks the arithmetic range of values. Simplest would be a min/max value for each node in the graph. This would be very useful in automatically narrowing arithmetic operations (e..g, if we know an unsigned multiply's operands are never larger than 255 (2**8 - 1) we know the result is always < 2**16 and so we can potentially make the multiplier smaller), as well as statically determining conditions.

The ranges would be propagated through operations. For example, if you know the lhs of an add is <= 100 and the rhs <= 200, then the result <= 300.

Initially it probably makes sense for the analysis to keep track of signed and unsigned ranges. A more general solution would keep track of a number of intervals that a value can be in.

cdleary, 2020-10-19:
A potentially interesting example here is specializing a function that takes many opcodes via a wrapper for a subset (e.g. two) of those opcodes: if we could track the two disjoint values we could cull all selects() so they only considered those two possibilities and DCE everything else.

cdleary, 2021-02-23:
Felix said he hadn't started on this quite yet so removing the lock for now.

taktoa, 2022-01-10:
For some reason I never marked this done

closed 2022-01-10 by taktoa