google/xls #164: Enumerate Operator not implemented in IR conversion

vincent-mirian-google, 2020-10-20:
When using the enumerate operator, the IR conversion tool throws an error.

Test case:
fn f(input: u32[4]) -> u32 {
  let result = 
    for ((i, elem), accum): ((u32, u32), u32) in **enumerate**(input) {
      u32:0
    }(u32:0);
  u32:0
}

Error:
~~~~~~~^---^ For-loop is of an unsupported form for IR conversion; only a 'range(0, const)' call is supported, found non-range callee.

To execute: ir_converter_main \<DSLX file with content above\>

hongted, 2020-10-21:
This issue is to address

https://github.com/google/xls/blob/2ebaec60e59e206df9491f00459f6936d481fb00/xls/dslx/ir_converter.py#L711-L715

allight, 2026-11-14:
Closing this issue as fixed (accepted by @erinzmoore): DSLX `for` loops now support arbitrary iterable expressions including range expressions `a..b`, array variables, and `enumerate(arr)` (`xls/dslx/ir_convert/function_converter.cc` and `xls/dslx/ir_convert/ir_converter_test.cc`), with direct tests in `xls/dslx/tests/for_over_array.x`. (Closed by Gemini)

closed 2026-11-14 by allight