google/xls #172: Read proto messages, derive code from it

dkillebrew-g, 2020-10-28:
For example, assume a sequence (`repeated`) of messages. Each message has integral `index`, `offset`, and `bitwidth`. Let's call each such message a "rule". The indices will be unique, and dense starting from 0. From these rules (e.g. let's say there are 100 of them) I'd like to create hardware. The hardware would do something like:

1. read an N-bit index from a buffer (N is `ceiling . log2` of the number of rules)
2. find the matching rule
3. read 32 additional bits from the buffer
4. write `bitwidth` bits (of the 32 bits just read) to `offset` of a second buffer

cdleary, 2021-02-19:
@RobSpringer any update for this issue?

RobSpringer, 2021-02-22:
Yeah! Sorry - this might be done, but I want to make sure all features have been implemented and that there's documentation before I mark it closed.

RobSpringer, 2021-02-22:
Yep! We're good to go. Closing!

closed 2021-02-22 by RobSpringer