google/xls #174: docker build fails

lromor, 2020-10-29:
Hi! 

I am unable to build and test using docker. 

```
docker build .
Sending build context to Docker daemon  22.43MB
Step 1/14 : FROM ubuntu:20.04
 ---> d70eaf7277ea
Step 2/14 : LABEL version="0.1"
 ---> Using cache
 ---> b8289016de89
Step 3/14 : LABEL description="Docker Image for Building/Testing XLS"
 ---> Using cache
 ---> 444d9e5232d4
Step 4/14 : RUN apt-get update -y
 ---> Using cache
 ---> 26ea9025813c
Step 5/14 : RUN apt-get install -y curl gnupg && curl -fsSL https://bazel.build/bazel-release.pub.gpg | gpg --dearmor > bazel.gpg && mv bazel.gpg /etc/apt/trusted.gpg.d/ && echo "deb [arch=amd64] https://storage.googleapis.com/bazel-apt stable jdk1.8" | tee /etc/apt/sources.list.d/bazel.list && apt-get update -y && apt-get install -y bazel
 ---> Using cache
 ---> 326ba645c34b
Step 6/14 : RUN apt-get -y install python3-distutils python3-dev python-is-python3 libtinfo5
 ---> Using cache
 ---> 1f24895e50b9
Step 7/14 : RUN apt-get install -y git vim
 ---> Using cache
 ---> ad75d7b4eb0d
Step 8/14 : RUN useradd -m xls-developer
 ---> Using cache
 ---> 0692c2b083e5
Step 9/14 : USER xls-developer
 ---> Using cache
 ---> 25cc6da61d39
Step 10/14 : WORKDIR /home/xls-developer/
 ---> Using cache
 ---> 52bfac6e8411
Step 11/14 : RUN git clone https://github.com/google/xls.git
 ---> Using cache
 ---> ee60e0361967
Step 12/14 : WORKDIR /home/xls-developer/xls
 ---> Using cache
 ---> bba0681754b2
Step 13/14 : RUN bazel build -c opt ...
 ---> Running in 183249d88aa1
Extracting Bazel installation...
Starting local Bazel server and connecting to it...
Loading: 
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded

```

Any ideas what could be the issue?

cdleary, 2020-10-30:
Hmm, that's surprising! It really stops at that point? No CPU activity?

Try adding the `-s` flag to the `bazel build` command in the docker file, that should create more verbose output. It does take a while because of some of the larger dependencies we have (LLVM, Z3), I'm wondering if it just looks like nothing is happening but it's churning?

Thanks for letting us know, I think we should be able to get to the bottom of this.

lromor, 2020-10-30:
My fault. Everything works fine.

closed 2020-10-30 by lromor