google/xls #177: Document codegen options available in the build_defs

cdleary, 2020-10-30:
For example, quick start should show how to plumb this:

```
dslx_test(
    name = "foo",
    srcs = ["foo.x"],
    deps = [
        "bar.x",
        "baz.x",
    ],
)

dslx_codegen(
    name = "foo_codegen",
    configs = [{
        "delay_model": "foo",
        "input_valid_signal": "input_valid",
        "output_valid_signal": "output_valid",
        "module_name": "my_foo_module",
        "pipeline_stages": 7,
        "reset": "rst",
        "reset_active_low": "false",
        "reset_asynchronous": "false",
    }],
    dslx_dep = ":foo",
)
```

cdleary, 2020-12-09:
The start of documentation for these is here: https://google.github.io/xls/build_system/#dslx_codegen

We should extend it, and also precisely describe a notion of flop layers vs stages in our terminology.

grebe, 2022-11-18:
I think this is closed by https://google.github.io/xls/codegen_options/, let me know if you disagree @cdleary.

closed 2022-11-18 by grebe