google/xls #184: Provide nice error message when test name is duplicated

dkillebrew-g, 2020-11-06:
### What happens now:

Input to XLS:

```
fn add(x: u32, y: u32) -> u32 {
  x + y + u32:0  // Something to optimize.
}

test add {
  assert_eq(add(u32:2, u32:3), u32:5)
}

test add {
  assert_eq(add(u32:2, u32:3), u32:6)
}
```

Command line:

`bazel run -c opt xls/dslx/interpreter:interpreter_main -- ~/add.x`

The error is a long stack trace with this misleading error message at the bottom: 

`KeyError: 'Could not find symbolic bindings for invocation: add(u32:2, u32:3)' `

### What should happen:

An error message that's more helpful, e.g. "test named 'add' has been duplicated. Please deduplicate"

cdleary, 2020-11-06:
Possible dup of https://github.com/google/xls/issues/181 -- will fix this one, note that we're transitioning the frontend from Python to C++ (so it can be easily invoked in C++-only environments) so frontend is slightly in flux, but having issues for these things is great, thanks for filing!

dkillebrew-g, 2020-11-06:
I do agree, they are similar. I'm not sure this is a dup. This one has a fatal error and stack trace, while #181 infinite loops.
Also, 181 is about test and function name collision, while this one is test and test name collision.

cdleary, 2021-02-19:
Not actively working on this at the moment so releasing back to the pool. It may already be creating an error message after the port to C++ but we'll want an `dslx/tests/errors/` test to demonstrate we handle it correctly and with a nice error message.

vincent-mirian-google, 2021-03-12:
Tested with:

fn add(x: u32, y: u32) -> u32 {
  x + y + u32:0  // Something to optimize.
}

#![test]
fn add() {
  assert_eq(add(u32:2, u32:3), u32:5)
}

#![test]
fn add() {
  assert_eq(add(u32:2, u32:3), u32:6)
}

Received error:
  0003: }
  0004: 
* 0005: #![test]
  ~~~~~~~~~^--^ ParseError: Test function 'add' has same name as module member @ issue_184.x:1:1-3:2
  0006: fn add() {
  0007:   assert_eq(add(u32:2, u32:3), u32:5)

Issue seems resolved. Can we close this issue?

vincent-mirian-google, 2021-03-12:
Tested with:
```
fn add(x: u32, y: u32) -> u32 {
  x + y + u32:0  // Something to optimize.
}

#![test]
fn add() {
  assert_eq(add(u32:2, u32:3), u32:5)
}

#![test]
fn add() {
  assert_eq(add(u32:2, u32:3), u32:6)
}
```
Received:
```
  0003: }
  0004: 
* 0005: #![test]
  ~~~~~~~~~^--^ ParseError: Test function 'add' has same name as module member @ issue_184.x:1:1-3:2
  0006: fn add() {
  0007:   assert_eq(add(u32:2, u32:3), u32:5)
```
Issue seems resolved. Can we close this issue?

vincent-mirian-google, 2021-03-12:
Seems like a new issue with referencing the function name in the message?

closed 2021-03-13 by copybara-service[bot]