google/xls #185: Tuple indices should be assumed to be non-negative integers

dkillebrew-g, 2020-11-06:
...so that we don't have to annotate tuple indices. IMO the explicit typing of the index is just noise.

Today we must do:

```
let t = (u32:2, u8:3);
assert_eq(u8:3, t[u32:1])
```

It would be better if we could do:

```
assert_eq(u8:3, t[1])
```

dkillebrew-g, 2020-11-09:
I'd like literals for shift amounts (right shift, left shift, etc.) to have the same interpretation. That is, literals should be inferred as non-negative integers (related note: are negative shift amounts valid?). This should work:

```
test shift_literal {
  let two = u2:0b10;
  assert_eq((two >> 1) as u1, u1:1)
}
```

cdleary, 2020-11-10:
Related to https://github.com/google/xls/issues/162 (but includes shift)

cdleary, 2020-11-10:
I think I'll hold off on shift amounts (just because they don't have to be compile time literal values) until we have unifying type inference instead of strictly deductive type inference. Would also apply to array indices and such at that point too.

cdleary, 2020-11-10:
So I will declare this blocked by issue #193 which I have now filed!

dkillebrew-g, 2020-11-11:
If you want to modify this bug (maybe make a note in #193 to ensure that shift amounts are inferred... but I'm sure they would be)  and declare this bug complete w.r.t. tuple indices, that makes sense to me.

cdleary, 2020-11-16:
Thanks, downscoped, made the note, closing!

closed 2020-11-16 by cdleary