google/xls #188: In-tree yosys binary doesn't have synth_{ecp,ice40} commands available

cdleary, 2020-11-09:
Need at least an `alwayslink=True` on the kernel link target and pointing somewhere else as the `YOSYS_DATDIR` in the defines. Been using out-of-tree binaries thus far, but we should be able to resolve this!

mithro, 2020-12-04:
FYI - @hzeller

mithro, 2020-12-04:
FYI - @QuantamHD

QuantamHD, 2020-12-04:
I think I've run into this issue before. I have a PR https://github.com/YosysHQ/yosys/pull/2417 out for it in yosys to allow you to set the datdir at runtime. Would that help?

QuantamHD, 2020-12-04:
You can also hack it together with something like this 

```
-DYOSYS_DATDIR='std::getenv(\"YOSYS_DATDIR\") ? std::getenv(\"YOSYS_DATDIR\") : \"{YOUR_DEFAULT_DATDIR_HERE}\"'
```

mithro, 2020-12-04:
https://github.com/masc-ucsc/livehd/blob/master/external/BUILD.yosys