google/xls #190: Bug: signed value is both negative and positive

dkillebrew-g, 2020-11-09:
Run this in the interpreter and it passes:

```
test widen_signed_cast {
  let negative_one = s2:0b11;
  let _ = assert_eq(negative_one, s2:-1);
  assert_eq(negative_one as s4, s4:3)
}

test numerical_conversions {
  let s8_m2 = s8:-2;
  // Sign extension (source type is signed).
  let _ = assert_eq(s32:-2, s8_m2 as s32);
  let _ = assert_eq(s16:-2, s8_m2 as s16);
  ()
}
```

IIUC, in the first test a negative value is zero extended to a positive value.
In the second test a negative value is sign extended to a wider version of the same negative value.

What's going on?

dkillebrew-g, 2020-11-09:
Actually, when the asserts pass but they should not, the full test file is this:
```
test narrow_signed_cast {
  let negative_seven = s4:0b1001;
  assert_eq(negative_seven as s2, s2:1)
}

test widen_signed_cast {
  let negative_one = s2:0b11;
  let _ = assert_eq(negative_one, s2:-1);
  assert_eq(negative_one as s4, s4:3)
}

test numerical_conversions {
  let s8_m2 = s8:-2;
  // Sign extension (source type is signed).
  let _ = assert_eq(s32:-2, s8_m2 as s32);
  let _ = assert_eq(s16:-2, s8_m2 as s16);
  ()
}
```
Interpret that file, and tests pass.

Now if we delete the first test, the widen_signed_cast test fails as we expect:

```
test widen_signed_cast {
  let negative_one = s2:0b11;
  let _ = assert_eq(negative_one, s2:-1);
  assert_eq(negative_one as s4, s4:3)
}

test numerical_conversions {
  let s8_m2 = s8:-2;
  // Sign extension (source type is signed).
  let _ = assert_eq(s32:-2, s8_m2 as s32);
  let _ = assert_eq(s16:-2, s8_m2 as s16);
  ()
}
```
failure message:
```
  0002:   let negative_one = s2:0b11;
  0003:   let _ = assert_eq(negative_one, s2:-1);
* 0004:   assert_eq(negative_one as s4, s4:3)
        ~~~~~~~~~~~^------------------------^ The program being interpreted failed! 
  lhs: bits[4]:0xf
  rhs: bits[4]:0x3
  were not equal
  0005: }
  0006: 
[           FAILED ] widen_signed_cast FailureError
```

cdleary, 2020-11-09:
There's some bug in test traversal it seems, likely introduced by the incremental factoring of the DSL to C++: if I put assert_eq(false, true) or trace a value in `widen_signed_cast` it says it runs but neither displays nor fails.

closed 2020-11-10 by copybara-service[bot]