google/xls #191: Confusing name/behavior for `signex`

dkillebrew-g, 2020-11-09:
Everyone is different, but for me personally, I find the behavior of `signex` doesn't match the name. For me, sign extend of an unsigned value shouldn't change its Real number interpretation, so unsigned types should be zero extended when "sign extending" them.  To be picky, unsigned values don't have a sign bit (they are un-signed or not signed), so there's no bit to duplicate.

Could we rename this primitive to `duplicate_msb` or `extend_msb` to reflect the fact that it's operating on the most significant bit? That is, the operation of this primitive has nothing to do with sign bits (or lack of sign bits). I think one of these names would be clearer to the set of users who share my interpretation of sign extension.

cdleary, 2020-11-10:
Any preferences on `msb_extend` vs `extend_msb`?

dkillebrew-g, 2020-11-10:
Lacking a pre-existing naming convention for this particular action (I Googled and found nothing), I have a mild preference for the way we'd say it in English prose: extend [the] msb. Verb subject. So `extend_msb`.

cdleary, 2021-02-19:
Not actively working on this ATM so releasing back to the pool and marking as good first bug -- we should also make sure we document it as a builtin per #306