google/xls #193: Add unifying type inference

cdleary, 2020-11-10:
Right now DSLX type inference is purely deductive (because it was simpler for proof of concept purposes). We should replace this with a unifying type inference algorithm (HM style) when the type inference is finished porting to C++ from Python.

cdleary, 2020-11-16:
Note per https://github.com/google/xls/issues/185#issuecomment-725637765 we'd want to make sure shift amounts are inferred with this completed (we expect they would be).

dkillebrew-g, 2021-01-15:
You suggested I add this use case as a motivation for this feature:

```
u32:2 as types::NumWidgets
```

This is quite a mouthful for the developer who really should just be able to type `2`!

Where the type is defined elsewhere, to be something like

```
pub type NumWidgets = uN[7];
```