google/xls #195: Add record types to IR

meheff, 2020-11-10:
A record type is a tuple with named fields. I'm guessing it will be possible to mostly just treat records as special tuples. Tupleindex instruction would work the same. Might need a new instruction kRecord which is analogous to kTuple but with field names. Similarly structured record types and tuple types would not be compatible.

The motivation is to enable richer types (structs) in the emitted SystemVerilog.

proppy, 2024-05-17:
@grotival was mentioning that this would enable the codegen to emit SystemVerilog w/ multiple output ports.

proppy, 2024-05-20:
https://internals.rust-lang.org/t/record-types/18258 discuss support for record type in Rust; which could be an inspiration for a future DSLX syntax.