google/xls #196: Support compile-time function evaluation of parameters

dkillebrew-g, 2020-11-11:
This would be useful:

```
fn bits_required(x : u32) -> u32 {
  x
}

#![test]
fn test_interpreter_import() {
  let _ = assert_eq(bits_required(u32:0), u32: 0);
  ()
}

struct Foo<NumThings : u32> {
  value: uN[bits_required(NumThings)]
}
```

Today it's a parse error:

```
* 0012:   value: uN[bits_required(NumThings)]
        ~~~~~~~~~~~~~~~~~~~~~~~~~^^ ParseError: platforms/darwinn/silo/rtl/xls/flit_to_delta.x:12:26-12:27 Expected ']', got '('
```

Also, it's a confusing error message if `bits_required` is defined in another module:

```
import math

#![test]
fn test_interpreter_import() {
  let _ = assert_eq(math::bits_required(u32:0), u32: 0);
  ()
}

struct Foo<NumThings : u32> {
  value: uN[math::bits_required(NumThings)]
}
```

`cpp_bindings.cc:60] Unsupported BoundNode variant: import math`

cdleary, 2020-11-18:
The fact the latter is a crash is an issue introduced by the port to C++ (and the lack of an easy open source pattern matcher for absl::variant).

RobSpringer, 2022-04-25:
Fixed by the big ol' constexpr overhaul I completed a while ago. A test demonstrating this is `constexpr_types_via_invocations.x`.

closed 2022-04-25 by RobSpringer