google/xls #198: Unhelpful error message when type alias is used

dkillebrew-g, 2020-11-11:
When I don't use a type alias, and my code is missing an annotation on the `0`, the error message is helpful:

```
struct Foo<FieldValueWidth : u32> {
  value : uN[2]
}

fn make_foo<FieldValueWidth : u32>
  () -> Foo<FieldValueWidth> {
    Foo{value : 0}
}

#![test]
fn test_make() {
  let s = make_foo<u32 : 1>();
  ()
}
```

```
* 0007:     Foo{value : 0}
        ~~~~~~~~~~~~~~~~^^ Could not infer type: Could not infer a type for this number, please annotate a type. @ foo.x:7:17-7:18
```

Let's use a type alias:

```
type Word = u32;

struct Foo<FieldValueWidth : Word> {
  value : uN[2]
}

fn make_foo<FieldValueWidth : Word>
  () -> Foo<FieldValueWidth> {
    Foo{value : 0}
}

#![test]
fn test_make() {
  let s = make_foo<Word : 1>();
  ()
}
```

Now the error message is not helpful at all, not even a line number to give us a clue what's the problem:

`AttributeError: 'xls.dslx.python.cpp_ast.TypeRe' object has no attribute 'bits'
`

dkillebrew-g, 2020-11-11:
Furthermore, I think type aliases are broken, beyond the error message. After properly annotating the value, the code, which I believe should work, is this:

```
type Word = u32;

struct Foo<FieldValueWidth : Word> {
  value : uN[2]
}

fn make_foo<FieldValueWidth : Word>
  () -> Foo<FieldValueWidth> {
    Foo{value : (uN[2] : 0)}
}

#![test]
fn test_make() {
  let s = make_foo<Word : 1>();
  ()
}
```

but I get the same error about `object has no attribute 'bits'`

Deleting the alias and replacing `Word` with `u32` causes the test to pass.

grebe, 2025-12-29:
I'm not sure when this got fixed (TSv2? before then?) but this seems to work for me now.