google/xls #201: Any possibility of compiling to Webassembly?

lachlansneff, 2020-11-16:
I'm not too familiar with Bazel, but it looks like it's possible to coerce it to compile a project to wasm. Something like LLVM really can't be compiled to wasm very easily, and a jit doesn't make sense, but would it be feasible to compile xls -> verilog parts to wasm?

cdleary, 2020-11-16:
Hi Lachlan, thanks for the q! I don't have too much experience with making WebAssembly-targeted C++ code bases, but it sounds like it's helpful that we /don't/ have a hard dependency on LLVM, it's just for fast native execution of the XLS program (we also have our IR interpreter as an alternative way of executing IR so it's possible to just chop out that dependency).

Our main non-stdlib "base" library is Abseil, it looks like this macro may be defined for when WebAssembly is targeted: https://abseil.io/docs/cpp/platforms/macros so could be some decent possibility?

cdleary, 2020-11-16:
If you were interested in trying to tackle the compatibility to see what it takes, you might consider starting with a target like `//xls/tools:codegen_main` which will take IR text and emit Verilog for it.

lachlansneff, 2020-11-16:
I've done some more investigation, and it looks like abseil only supports compiling some parts of itself to target emscripten, which is may or may not run in standalone wasm runtimes depending on if it uses few enough system calls to target wasi or not.

cdleary, 2020-11-17:
Thanks for letting us know. We're generally going to be dependent on our base library, so sounds like it probably comes down to whether it ends up using few enough syscalls, you're saying?

cdleary, 2020-12-15:
Since I'm not sure we have something we can actively do on our side for this one, I'm thinking closing this issue in a few days seems right, feel free to file followups if there are specific things that we can do to help!

lachlansneff, 2020-12-15:
Absolutely. When I have time, I'll try to get this working, and then I'll reopen or open a new issue/pr.

closed 2020-12-15 by lachlansneff

cdleary, 2020-12-15:
Thanks Lachlan!