google/xls #202: Ubuntu 18.04: undeclared inclusion(s) in rule '@com_google_absl//absl/base:log_severity'

lachlansneff, 2020-11-16:
Here's the error:

```
❯ bazel build //xls/tools:codegen_main
INFO: Analyzed target //xls/tools:codegen_main (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
ERROR: /home/lachlan/.cache/bazel/_bazel_lachlan/d8365cac006c3b38c2028f7288f6928e/external/com_google_absl/absl/base/BUILD.bazel:54:11: undeclared inclusion(s) in rule '@com_google_absl//absl/base:log_severity':
this rule is missing dependency declarations for the following files included by 'com_google_absl/absl/base/log_severity.cc':
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stddef.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stdarg.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/stdint.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/limits.h'
  '/usr/lib/gcc/x86_64-linux-gnu/9/include/syslimits.h'
Target //xls/tools:codegen_main failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 3.369s, Critical Path: 3.01s
INFO: 13 processes: 13 internal.
FAILED: Build did NOT complete successfully
```

I'm using ubuntu running within wsl.

cdleary, 2020-11-17:
Since that target comes from @com_google_absl I think this needs to be filed against the absl repo? https://github.com/abseil/abseil-cpp

cdleary, 2020-11-17:
(May want to sanity check some binary that uses the basic facilities gets the same issue, and whether there's documentation on WebAssembly build flags for absl?)

lachlansneff, 2020-11-17:
This was actually an error that resulted from compiling to x86_64 on macos.

cdleary, 2020-11-17:
I don't have access to WSL to repro. Any chance you use Docker with our provided Dockerfile?

lachlansneff, 2020-11-17:
Compiling works fine when running the docker file, so this isn't a blocker, but it's unclear to me whether this is a dependency  issue or whether the abseil doesn't like compiling in WSL.

g-easy, 2020-11-30:
Try `bazel clean --expunge`

cdleary, 2020-12-22:
This might be more an issue for Abseil than XLS perhaps? Will close in a few days if we don't think it's actionable on the XLS side, thanks for letting us know though!

closed 2020-12-31 by cdleary