google/xls #203: Please add warning for "defined but did not use"

dkillebrew-g, 2020-11-16:
Please add a warning when an identifier is bound to a value, but never used (e.g. in a pattern match). It would help prevent silly programmer errors like the following, where I forgot to prefix my enum values with `Validity::`

```
enum Validity : u1 {
  Invalid = 0,
  Valid = 1
}

fn apply_differences(v : Validity) -> bool {
    match v {
      Valid => true,
      Invalid => false,
      _ => false,
  }
}

#![test]
fn t() {
  let _ = assert_eq(apply_differences(Validity::Valid), true);
  let _ = assert_eq(apply_differences(Validity::Invalid), false);
  ()
}
```

Right now, the interpreter (correctly) tells me that the second assert fails. What I want is for the compiler to warn me that the match has two defined but unused values.

cdleary, 2021-02-19:
This would be a good first bug for folks interested in the frontend, we have def/use chains on name definitions, could potentially be a post-pass after parsing to emit warnings fornames that were defined with no associated uses.

cdleary, 2023-08-17:
Update note: code is written, just getting the world compliant before landing it (since we use warnings-as-errors by default)...

closed 2023-08-22 by copybara-service[bot]