google/xls #205: Please add sharing (i.e. export and import) of constants between modules

dkillebrew-g, 2020-11-17:
An example:

definition.x
```
// right now, can't use `pub`
pub const FOO =  u32 : 1;
```

importer.x
```
import definition

// right now, both of these are an error. Of course, we'd choose one of these syntaxes and delete the other line...
const IMPORTED_FOO = definition::FOO;
type IMPORTED_FOO = definition::FOO;
```

I don't know which syntax rust uses for this kind of thing (const, or type, or something else). I was just putting up a strawman example.

cdleary, 2020-11-30:
Some support was added in https://github.com/google/xls/commit/581039158f4c6c6e9d6380994014513385d610c9 -- but not for recognizing colon-referenced entities as being constants, which is in your `importer.x` example, that's still outstanding here.

cdleary, 2021-02-25:
I believe this is done, e.g. see https://github.com/google/xls/blob/main/xls/dslx/tests/import_constexpr.x#L20 -- looks like Rob removed the restriction in https://github.com/google/xls/commit/0d8d9718698b7328320a863f42bc6aaa68ee0d5c#diff-8dec34b2e2bd2ccb0ba856c686d64235e0bc14ec4ecb03174a92309ba3214613

closed 2021-02-25 by cdleary