google/xls #207: Constant is not usable in slice expression

dkillebrew-g, 2020-11-17:
I have a constant that is being rejected by a slice expression:

```
const SIZE_BITS = u32 : 1000;
type LotsOfBits = uN[SIZE_BITS];

fn doit(x : u32, lotsabits : LotsOfBits)
  -> LotsOfBits {
    lotsabits[:10] ++ x ++ lotsabits[42:SIZE_BITS]
}
```

the error message is
```
  0004: fn doit(x : u32, lotsabits : LotsOfBits)
  0005:   -> LotsOfBits {
* 0006:     lotsabits[:10] ++ x ++ lotsabits[42:SIZE_BITS]
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^------------^ ParseError: file.x:6:37-6:51 Only constant numbers are currently allowed in slice expressions.
  0007: }
```

Of course this doesn't make sense because `SIZE_BITS` is a constant.

taktoa, 2023-02-28:
Reopen because the bug was not fixed, we just added a regression test

cdleary, 2023-08-10:
Slightly confused about the history here -- I think the regression test demonstrated that this was fixed? Constants are now usable in slice expressions and we've got that regression test for it.

closed 2023-08-10 by cdleary