google/xls #208: Allow different types for shift operators

dkillebrew-g, 2020-11-19:
In DSLX, please lift the restriction that shift_amount must be the same type as shifted_value. 
Motivation: the current restriction causes a bunch of extraneous casting in my code.

cdleary, 2021-02-19:
This would be a good first bug for somebody looking to get started in the DSLX frontend. Would need to define new type inferencing rules for shift operators (right now they're treated as `(T, T) -> T` signature like other arithmetic binops).

vincent-mirian-google, 2021-03-02:
In the following example, what should the deduced type be? sN[16]? a casted value?
let x = sN[16]:1000;
let y = sN[32]:-1000;
x << y

Thanks.

dkillebrew-g, 2021-03-02:
Vincent, based on your example, I'm not sure you understood the request. Are you asking what the type of `y` should be? Your example specifies that `y`, the shift amount, is `sN[32]`, so your example is unaffected by the request.

Anyway, if I were to provide an illustrative example:

```
let x = sN[32]:<some runtime-computed value>;
x << 5
```

The request is that the above type checks, and that `5` is inferred to have a type. The type inference of the expression's result is not addressed by this request, that's orthogonal.

There are a number of ways to solve this. Here's my initial solution: 
`5` should be inferred to be the "smallest representative type". In this case, 5 is non-negative, so let's use an unsigned type. `clog2(5)`  is `3`, so we choose `uN[3]`.

closed 2021-03-12 by copybara-service[bot]