google/xls #209: Allow explicit conversion between struct and bits (i.e. marshal and unmarshal)

dkillebrew-g, 2020-11-20:
For example, this would be useful: 

```
#![packed ascending order]
struct Packet {
  // occupies the least significant 5 bits, when converted via `as`
  header : uN[5],
  // occupies the most significant 20 bits, when converted via `as`
  body : uN[20]
}

fn make_packet(bits : uN[25]) -> Packet {
  bits as Packet
}
fn deconstruct_packet(p : Packet) -> bits : uN[25] {
  p as uN[25]
}
```

Motivation: when interfacing DSLX with existing code (e.g. Verilog), data comes in, and goes out as raw bits. Sometimes the incoming/outgoing bits includes multiple related, but different, pieces of data packed together into a single bits. When the data enters DSLX land, it should be typed. Outgoing types should be converted to bits. I think bitcasting between a struct and bits of the exact same bitwidth would be a much more ergonomic interface than the alternative, where the developer must do lots of shifting, masking, ORing, etc. This feature would go a long way to encouraging developers to do the right thing and type their data, provide descriptive names, etc.

Under the hood, `bits as Packet` would desugar to some shifting and truncating (to the width of each struct field). `p as uN[25]` would desugar to concatenation. 

An attribute would let the developer opt-in on a struct-by-struct basis (so that we're not 'breaking' the type system willy-nilly). 

The attribute could also provide control over how fields are packed (assuming more than one way of packing is useful). `packed ascending order` is a strawman example, I'm not particularly attached to it. Maybe with the addition of `packed descending order` we'd capture most use cases.

mikex-oss, 2024-12-05:
+1. I'll note that codegen turns structs/tuples into flattened bit vectors with an assumed ordering that the user must know to interface with the XLS-generated module. 

So `p as uN[25]` doesn't seem that surprising. 

Also saw #1698 which seems to be an alternative to generate ephemeral functions instead of `as` casting.

dank-openai, 2025-02-27:
Today I wanted to `one_hot_select` an array of things. The things were not simple bits types (one_hot_select requires the argument to be bits). They were tuples, each tuple element a struct of a different type. It would have been nice if 

- the compiler provided me marshal/unmarshal (AKA flatten/unflatten) functions for my tuple of things so that I could use `one_hot_select`
-  OR`one_hot_select` could work on non-bits types

allight, 2025-02-27:
Honestly the second (OHS takes any value type) is pretty easy (in the opt/codegen pipeline at least) and I think a much better solution to this problem.

Coercing values to bits can make many optimizations harder to implement or perform since the struct/tuple representation has ready-made split points between sequences of related bits. This makes recognizing and doing things like narrowing much easier.

ericastor, 2025-02-27:
I'm not convinced that `one_hot_select` should be used for arbitrary types, where there's no obvious equivalent of "bitwise OR". Also, OHS can prevent many optimizations, since the output isn't guaranteed to be exactly equal to one of the inputs!

@dank-openai, can you share a bit more about the application you have in mind, and why `one_hot_select` is more appropriate than (e.g.) `priority_select`? (Also, XLS will generally codegen priority select with the same Verilog as OHS if we can prove that the selector is always one-hot!)

allight, 2025-02-27:
I mean I'd think we'd do it as an OR of the leaf types.

EG

```
OHS(sel, Tuple[A, B, C], Tuple[D, E, F], Tuple[X, Y, Z])
```

is the same as

```
Tuple[OHS(sel, A, D, X), OHS(sel, B, E, Y), OHS(C, F, Z)]
```

That's effectively what is being asked for in any case.

No argument about OHS usually being the wrong tool to reach for though.

dank-openai, 2025-03-10:
@allight , @ericastor , thanks for looking at the issue, your consideration is appreciated. Apologies for the delay getting back to you.

> can you share a bit more about the application you have in mind, and why one_hot_select is more appropriate than (e.g.) priority_select

Sure, the essence is:

- I have a non-power of two number of these tuples, and I'm selecting one of them. They are constants.
- There's an `input : u10` that will determine which is selected. However, sometimes I use only 9 of the bits to select, other times 10. The array of tuples is zipped with an array that tells me if I should match using 9 or 10 bits, and zipped with an array that is the match pattern. 
- The logic is: based on the "9 or 10" flag, test if the given match pattern and `input`'s 9 or 10 bits are equal. Now I have an array of booleans.
- I know that the match patterns and the "9 or 10 bits matched?" are, taken as a whole, all mutually exclusive. Thus, `priority_sel` is inappropriate. Also I don't want a default value, if the default was ever chosen, that would be a logic error that I want to find during testing. So the array of booleans (generated in the step above) will be one hot.
- Now with my one hot boolean array, I want to select the appropriate tuple.

Now you might be asking why I didn't use a `for` loop, pass the loop a default selection, iterate over the many tuples, and return the 'last' of the tuples that matched `input` (which would be logically equivalent to a priority select, IIUC). Well, as stated above, a default selection would be an error that I want to know about. Additionally, it's my understanding that OHS is more optimal than other options, e.g. iterating. Of course I want optimal code generation here, as I know these options are all mutually exclusive. I don't expect XLS to perform the advanced(?) and expensive optimization logic required to determine that my many match patterns and associated "9 or 10 bits matched?" are, taken as a whole, all mutually exclusive. I don't expect it to prove to itself that exactly one of the tuples will match, not 0, not more than 1, and thus generate OHS on its own.

So it boils down to: 

- I know my cases are mutually exclusive. 
- I want (need?) the most optimal code generation. 
- I strongly doubt that XLS will be able to prove this to itself. Chris tells me the only way to guarantee that XLS knows this is for me to invoke OHS myself. 
- Finally, creating the OHS selector was fairly elegant, a map and zip and `repeat` and a 5 line "does this match?" function. I think it's a better way of expressing the logic than a `for` loop or similar, which is not exactly the same semantics, as I don't want a default case or priority selection.

Let me know if there's any clarification I can provide.

allight, 2025-03-10:
As long as the one-hot-ness of the selector (ie Mutex) is provable by XLS (which can be forced by just wrapping it in a [`std::one_hot`](https://google.github.io/xls/dslx_std/#one_hot)) a priority select will have exactly the same QOR as a one-hot-select (in fact they will be codegen'd identically since a [codegen pass transforms priority-sels to OHS if the input is one-hot](https://github.com/google/xls/blob/e4e13be78c142cd3d1876b728c70034a04857325/xls/codegen/priority_select_reduction_pass.cc#L77)).

Just generally OHS without a trivial one-hot-ness/mutex proof (like a std::one_hot select) is a pretty big opt barrier since it increases uncertainty about the nodes final value.

I'd just check and make sure that using a priority select is actually a QoR problem.

ericastor, 2025-03-10:
Also, if you want to fail if the array is not in fact one-hot... that's easy enough to add to the approach Alex outlined above! Just assert that the `std::one_hot` version's leading bit (which signals that no input was set) is in fact zero, and then pass all other bits as the selector to a priority select.

Also, you could consider using a match statement, rather than invoking priority select directly. If you'd like to request a new match statement that expresses "exactly one of these conditions should match, and otherwise we should `fail!`", it could be a nice bit of syntactic sugar for some use cases... and we can compile it (in the background) to what Alex described.

dank-openai, 2025-03-13:
To circle back to the original customer pain, let's say I have an array of booleans (to select a value), and a corresponding array of non-bits (i.e. more complex types) values. Can we make it easy to use OHS and `priority_sel` to select a value?

ericastor, 2025-03-14:
Thanks for focusing back on the goal, rather than the method! I don't think this issue is the right place for further discussion on this topic, since the conversion between struct & bits is not what we'd recommend as a solution to these needs.

I've opened a discussion here: https://github.com/google/xls/discussions/2002