google/xls #211: Fix parsing of constant array of structures

dkillebrew-g, 2020-11-20:
This code does not parse, but should:

```
struct FooAndBar {
  foo : u32,
  bar : u32
}

const MY_FOO_BARS = FooAndBar[2] :
  [
    FooAndBar{foo:1,bar:2}
  , FooAndBar{foo:3,bar:4}
  ];
```

Current error message:

```
google3.third_party.pybind11_abseil.status.StatusNotOk: INVALID_ARGUMENT: ParseError: foobar.x:7:3-10:4 Value is not considered constant: '[struct FooAndBar {
  foo: u32,
  bar: u32,
} { foo: 1, bar: 2 }, struct FooAndBar {
  foo: u32,
  bar: u32,
} { foo: 3, bar: 4 }]'
```

cdleary, 2021-01-05:
I think Rob's new constant expression support should handle this so assigning to him just to make sure he gets test cases in for this (if so).

RobSpringer, 2022-04-25:
Yup. I'll add a unit test and close this one.

closed 2022-04-26 by RobSpringer