google/xls #214: Parametric value not usable in slice expression

dkillebrew-g, 2020-11-20:
I understand "constexpr" is a work in progress. Just letting you know this case is useful, but not parsing yet:

```
fn pad<X : u32, R : u32, DIFF : u32 = R-X>(xs : u32[X]) -> u32[R] {
  xs ++ u32[DIFF] : [0, ...]
  if R >= X
  else xs[:R]
}
```

So you can use it as a test case when you're done with constexpr work.

Currently, the error is

```
* 0167:   else xs[:R]
        ~~~~~~~~~^--^ ParseError: blah.x:167:10-167:14 Only constant numbers are currently allowed in slice expressions.
```

cdleary, 2021-01-22:
Just a note that #207 is very related, both all about constexpr evaluation.

cdleary, 2021-03-08:
I think I failed to realize on my first reading this is actually an *array* slice, not a *bit* slice. So blocked on https://github.com/google/xls/issues/312 and https://github.com/google/xls/issues/126

closed 2023-06-23 by cdleary