google/xls #223: JIT/interpreter miscompare, crasher 8671

meheff, 2020-12-05:
Miscompare repros outside of JIT, but minimized test case is large.

[crasher_8671.tar.gz](https://github.com/google/xls/files/5647761/crasher_8671.tar.gz)

cdleary, 2021-01-05:
Do you think the minimizer needs work here @meheff before we take some action?

meheff, 2021-01-05:
Potentially. One could try to minimize the sample by hand and then port any such transformations to the minimizer. It would likely have to be some kind of complicated transform as the minimizer likely does most of the easy stuff already.

meheff, 2021-01-05:
btw, here's what the IR looks like:

package sample

fn __sample__main(x0: bits[24], x1: bits[39]) -> (bits[1][39], bits[24], bits[45], bits[39], bits[39], bits[39], bits[1][117], bits[1][78], bits[24], bits[1][39], bits[45], bits[1][117], bits[39], bits[39], bits[39], bits[39], bits[1][39], bits[39], bits[1][117]) {
  zero_ext.50: bits[39] = zero_ext(x0, new_bit_count=39, id=50, pos=0,8,19)
  x9: bits[39] = add(zero_ext.50, x1, id=51, pos=0,8,32)
  bit_slice.41: bits[1] = bit_slice(x1, start=38, width=1, id=41)
  bit_slice.40: bits[1] = bit_slice(x1, start=37, width=1, id=40)
  bit_slice.39: bits[1] = bit_slice(x1, start=36, width=1, id=39)
  bit_slice.38: bits[1] = bit_slice(x1, start=35, width=1, id=38)
  bit_slice.37: bits[1] = bit_slice(x1, start=34, width=1, id=37)
  bit_slice.36: bits[1] = bit_slice(x1, start=33, width=1, id=36)
  bit_slice.35: bits[1] = bit_slice(x1, start=32, width=1, id=35)
  bit_slice.34: bits[1] = bit_slice(x1, start=31, width=1, id=34)
  bit_slice.33: bits[1] = bit_slice(x1, start=30, width=1, id=33)
  bit_slice.32: bits[1] = bit_slice(x1, start=29, width=1, id=32)
  bit_slice.31: bits[1] = bit_slice(x1, start=28, width=1, id=31)
  bit_slice.30: bits[1] = bit_slice(x1, start=27, width=1, id=30)
  bit_slice.29: bits[1] = bit_slice(x1, start=26, width=1, id=29)
  bit_slice.28: bits[1] = bit_slice(x1, start=25, width=1, id=28)
  bit_slice.27: bits[1] = bit_slice(x1, start=24, width=1, id=27)
  bit_slice.26: bits[1] = bit_slice(x1, start=23, width=1, id=26)
  bit_slice.25: bits[1] = bit_slice(x1, start=22, width=1, id=25)
  bit_slice.24: bits[1] = bit_slice(x1, start=21, width=1, id=24)
  bit_slice.23: bits[1] = bit_slice(x1, start=20, width=1, id=23)
  bit_slice.22: bits[1] = bit_slice(x1, start=19, width=1, id=22)
  bit_slice.21: bits[1] = bit_slice(x1, start=18, width=1, id=21)
  bit_slice.20: bits[1] = bit_slice(x1, start=17, width=1, id=20)
  bit_slice.19: bits[1] = bit_slice(x1, start=16, width=1, id=19)
  bit_slice.18: bits[1] = bit_slice(x1, start=15, width=1, id=18)
  bit_slice.17: bits[1] = bit_slice(x1, start=14, width=1, id=17)
  bit_slice.16: bits[1] = bit_slice(x1, start=13, width=1, id=16)
  bit_slice.15: bits[1] = bit_slice(x1, start=12, width=1, id=15)
  bit_slice.14: bits[1] = bit_slice(x1, start=11, width=1, id=14)
  bit_slice.13: bits[1] = bit_slice(x1, start=10, width=1, id=13)
  bit_slice.12: bits[1] = bit_slice(x1, start=9, width=1, id=12)
  bit_slice.11: bits[1] = bit_slice(x1, start=8, width=1, id=11)
  bit_slice.10: bits[1] = bit_slice(x1, start=7, width=1, id=10)
  bit_slice.9: bits[1] = bit_slice(x1, start=6, width=1, id=9)
  bit_slice.8: bits[1] = bit_slice(x1, start=5, width=1, id=8)
  bit_slice.7: bits[1] = bit_slice(x1, start=4, width=1, id=7)
  bit_slice.6: bits[1] = bit_slice(x1, start=3, width=1, id=6)
  bit_slice.5: bits[1] = bit_slice(x1, start=2, width=1, id=5)
  bit_slice.4: bits[1] = bit_slice(x1, start=1, width=1, id=4)
  bit_slice.3: bits[1] = bit_slice(x1, start=0, width=1, id=3)
  zero_ext.104: bits[40] = zero_ext(x9, new_bit_count=40, id=104, pos=0,12,20)
  x2: bits[1][39] = array(bit_slice.41, bit_slice.40, bit_slice.39, bit_slice.38, bit_slice.37, bit_slice.36, bit_slice.35, bit_slice.34, bit_slice.33, bit_slice.32, bit_slice.31, bit_slice.30, bit_slice.29, bit_slice.28, bit_slice.27, bit_slice.26, bit_slice.25, bit_slice.24, bit_slice.23, bit_slice.22, bit_slice.21, bit_slice.20, bit_slice.19, bit_slice.18, bit_slice.17, bit_slice.16, bit_slice.15, bit_slice.14, bit_slice.13, bit_slice.12, bit_slice.11, bit_slice.10, bit_slice.9, bit_slice.8, bit_slice.7, bit_slice.6, bit_slice.5, bit_slice.4, bit_slice.3, id=42, pos=0,2,23)
  encode.56: bits[6] = encode(zero_ext.104, id=56)
  x8: bits[1][78] = array_concat(x2, x2, id=49, pos=0,7,26)
  x13: bits[39] = zero_ext(encode.56, new_bit_count=39, id=57, pos=0,12,20)
  literal.100: bits[45] = literal(value=0, id=100, pos=0,20,21)
  x12: bits[1][117] = array_concat(x8, x2, id=54, pos=0,11,27)
  x10: bits[1][117] = array_concat(x2, x8, id=52, pos=0,9,27)
  ret tuple.74: (bits[1][39], bits[24], bits[45], bits[39], bits[39], bits[39], bits[1][117], bits[1][78], bits[24], bits[1][39], bits[45], bits[1][117], bits[39], bits[39], bits[39], bits[39], bits[1][39], bits[39], bits[1][117]) = tuple(x2, x0, literal.100, x9, x9, x13, x12, x8, x0, x2, literal.100, x12, x1, x1, x9, x1, x2, x1, x10, id=74, pos=0,27,2)
}

cdleary, 2021-01-05:
What if the mutator randomly rewrote a use to another use of the same type in the available ops? Was wondering if we don't have that one yet.

meheff, 2021-01-05:
That would be an interesting one. Would have to track reachability in some way to avoid adding cycles. Toposort could be a crude approximation.

vincent-mirian-google, 2022-12-10:
Add top keyword to crasher options . New crasher below:
```
// options: {"input_is_dslx": true, "ir_converter_args": ["--top=main"], "convert_to_ir": true, "optimize_ir": true, "use_jit": true, "codegen": true, "codegen_args": ["--use_system_verilog", "--generator=combinational"], "simulate": false, "simulator": null, "use_system_verilog": true}
// args: bits[24]:0x7f_ffff; bits[39]:0x3f_8eff_4082
// args: bits[24]:0xaa_aaaa; bits[39]:0x1
// args: bits[24]:0x0; bits[39]:0x10
// args: bits[24]:0x7f_ffff; bits[39]:0x2000_0000
// args: bits[24]:0xaa_aaaa; bits[39]:0x80
// args: bits[24]:0x4_0000; bits[39]:0x2_0000_0080
// args: bits[24]:0x8; bits[39]:0x12_0004_5555
// args: bits[24]:0x800; bits[39]:0x10_b497_4b8a
// args: bits[24]:0x7f_ffff; bits[39]:0x1_0000_0000
// args: bits[24]:0x4_0000; bits[39]:0x4
// args: bits[24]:0xb4_2eaf; bits[39]:0x2000
// args: bits[24]:0x2000; bits[39]:0x24_5000_1c80
// args: bits[24]:0x4000; bits[39]:0xa280_2500
// args: bits[24]:0x80_0000; bits[39]:0x40_4000_37e6
// args: bits[24]:0x10; bits[39]:0x8_0000_0000
// args: bits[24]:0xbe_3a17; bits[39]:0x80_0000
// args: bits[24]:0x2000; bits[39]:0x80
// args: bits[24]:0x36_3707; bits[39]:0x53_0393_d042
// args: bits[24]:0x0; bits[39]:0x2000
// args: bits[24]:0x8; bits[39]:0x8_0000_0000
// args: bits[24]:0x1_0000; bits[39]:0x6_80a2_4545
// args: bits[24]:0x800; bits[39]:0x9_5374_6105
// args: bits[24]:0x400; bits[39]:0x800_0000
// args: bits[24]:0x8; bits[39]:0x40_0806_0104
// args: bits[24]:0x10_0000; bits[39]:0x48_1599_0c56
// args: bits[24]:0x0; bits[39]:0x40_0000_0000
// args: bits[24]:0x400; bits[39]:0x21_5602_4021
// args: bits[24]:0x2000; bits[39]:0xc_3007_0840
// args: bits[24]:0x4_0000; bits[39]:0x30_c444_b42d
// args: bits[24]:0x4_0000; bits[39]:0x42_02d0_36a8
// args: bits[24]:0x100; bits[39]:0x2_4092_9761
// args: bits[24]:0x400; bits[39]:0x8_0240_0edc
// args: bits[24]:0xcb_64ed; bits[39]:0x45_6356_bb78
// args: bits[24]:0x80_0000; bits[39]:0x1000_0000
// args: bits[24]:0x10_0000; bits[39]:0xe_6c38_4572
// args: bits[24]:0x80; bits[39]:0x40_0101
// args: bits[24]:0x20; bits[39]:0x4
// args: bits[24]:0x70_8f30; bits[39]:0x6a_e28c_7080
// args: bits[24]:0x7f_ffff; bits[39]:0x4_0000_0000
// args: bits[24]:0x20_0000; bits[39]:0x10_c409_7bff
// args: bits[24]:0x8000; bits[39]:0x4_0000_0000
// args: bits[24]:0x40; bits[39]:0x20
// args: bits[24]:0x80; bits[39]:0x2
// args: bits[24]:0x800; bits[39]:0x1_0000_0000
// args: bits[24]:0x10_0000; bits[39]:0xa_41ac_0a1c
// args: bits[24]:0x2000; bits[39]:0x40_1000_5556
// args: bits[24]:0x8_0000; bits[39]:0x5a_48b3_ec67
// args: bits[24]:0x55_5555; bits[39]:0x2a_a8eb_aaaa
// args: bits[24]:0x80_0000; bits[39]:0x200_0000
// args: bits[24]:0x40; bits[39]:0x40
// args: bits[24]:0x20; bits[39]:0x80_0000
// args: bits[24]:0x80; bits[39]:0x2440_6226
// args: bits[24]:0x1; bits[39]:0x32_2581_682c
// args: bits[24]:0xba_288e; bits[39]:0x4000
// args: bits[24]:0x80_0000; bits[39]:0x4_0000_0000
// args: bits[24]:0x0; bits[39]:0x1_0000
// args: bits[24]:0xaa_aaaa; bits[39]:0x3f_ffff_ffff
// args: bits[24]:0xff_ffff; bits[39]:0x4_0000_0000
// args: bits[24]:0x55_5555; bits[39]:0x1a_ba09_eb7b
// args: bits[24]:0x2_0000; bits[39]:0x1_0008_0c11
// args: bits[24]:0xa5_2ff6; bits[39]:0x1
// args: bits[24]:0x2f_57cb; bits[39]:0x12_8140_f4a4
// args: bits[24]:0x2_0000; bits[39]:0x800_0000
// args: bits[24]:0x7f_ffff; bits[39]:0x200
// args: bits[24]:0x2; bits[39]:0x30_081f_004c
// args: bits[24]:0x100; bits[39]:0x80_0000
// args: bits[24]:0x8; bits[39]:0x2_1005_0022
[... 73 lines cut ...]
  let x13: u39 = ctz(x9);
  let x14: u39 = for (i, x): (u4, u39) in range(u4:0x0, u4:0x7) {
    x
  }(x5);
  let x15: u39 = for (i, x): (u4, u39) in range(u4:0x0, u4:0x0) {
    x
  }(x9);
  let x16: (u24, u39) = (x0, x9);
  let x17: u45 = u45:0x20000;
  let x18: u39 = rev(x1);
  let x19: u45 = x17;
  let x20: u39 = for (i, x): (u4, u39) in range(u4:0x0, u4:0x1) {
    x
  }(x1);
  let x21: u39 = (x13) ^ (x15);
  (x2, x0, x19, x9, x21, x13, x12, x8, x0, x2, x17, x12, x1, x1, x15, x5, x2, x18, x10)
}
```

Success with:
```
blaze run -c opt //xls/fuzzer:run_crasher -- sample.x 
```

Closing as fixed.