google/xls #224: JIT/Interpreter miscompare, crasher c89a

meheff, 2020-12-08:
Looks like a select with a non-selected poison operand ultimately gets replaced with poison. Effectively this results in the result not being stored in the result buffer (store of poison gets removed by llvm?).

https://bugs.llvm.org/show_bug.cgi?id=48435

Potential workaround might be to guard overshifted values using if/then block rather than select depending on resolution of bug.

[crasher_c89a.tar.gz](https://github.com/google/xls/files/5656381/crasher_c89a.tar.gz)

cdleary, 2021-01-05:
I think this is blocked on the LLVM issue which is still not yet resolved.

meheff, 2021-01-05:
Might be a while before that LLVM bug is fixed as these poison problems are pretty pervasive and fixing them has performance implications. We might want to change the way we handle shifts in the meantime. Instead of using selects we should try if/then/else blocks. That may be less likely to run afoul of poison handling problems in LLVM.

cdleary, 2021-01-06:
Sounds good to me, I guess we should do sooner rather than later if it can produce miscompiles. We'll have to figure out who can pick it up, @hongted @vincent-mirian-google could be interesting one to pick up if you want to learn more about the JIT and the LLVM IR builder API! (But we probably need to hop on it immediately given it can hypothetically cause faulty verification.)

cdleary, 2021-01-06:
We said in a brief meeting today that @meheff was going to take this since it can lead to corruption if LLVM optimizes incorrectly. Thanks Mark!

commit c4b6934d (2021-01-07): Avoid creating poisonous shifts in LLVM JIT.
LLVM sometimes improperly optimizes overshifted shifts which create poisonous values. Change the code generation around shifts to avoid these cases.

Fixes #224.

PiperOrigin-RevId: 350593522
closed 2021-01-07 by copybara-service[bot]