google/xls #225: JIT duplicates body of non-entry functions

meheff, 2020-12-09:
When a function has multiple callers the body is duplicated in the JIT. Example:define i1 @test(i7 %x0, i64 %_2) {
so_basic:
  %invoke_198 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 16777216, i64 %_2)
  %invoke_200 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 %invoke_198, i64 %_2)
  %invoke_202 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 %invoke_200, i64 %_2)
  %invoke_204 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 %invoke_202, i64 %_2)
  %invoke_206 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 %invoke_204, i64 %_2)
  %x5 = call i27 @"sample::____sample__main_counted_for_0_body"(i4 0, i27 %invoke_206, i64 %_2)
...
}

define i27 @"sample::____sample__main_counted_for_0_body"(i4 %i, i27 %x, i64 %_0) {
so_basic:
  ret i27 %x

so_basic1:                                        ; No predecessors!
  ret i27 %x

so_basic2:                                        ; No predecessors!
  ret i27 %x

so_basic3:                                        ; No predecessors!
  ret i27 %x

so_basic4:                                        ; No predecessors!
  ret i27 %x

so_basic5:                                        ; No predecessors!
  ret i27 %x

so_basic6:                                        ; No predecessors!
  ret i27 %x

so_basic7:                                        ; No predecessors!
  ret i27 %x

so_basic8:                                        ; No predecessors!
  ret i27 %x

so_basic9:                                        ; No predecessors!
  ret i27 %x

so_basic10:                                       ; No predecessors!
  ret i27 %x

so_basic11:                                       ; No predecessors!
  ret i27 %x
}

cdleary, 2021-01-05:
JIT bug so assigning @RobSpringer

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `JittedFunctionBase::BuildInternal` (`xls/jit/function_base_jit.cc:1359-1382`) now compiles each transitively dependent `FunctionBase` once in topological order (asserting no duplicate `llvm::Function` is created) and caches it via `SetLlvmFunction`, which `IrBuilderVisitor::GetFunction` (`xls/jit/ir_builder_visitor.cc:1452`) reuses across all `Invoke`, `CountedFor`, and `Map` call sites. (Closed by Gemini)

closed 2026-11-14 by allight