google/xls #230: Add "constexpr function" support

dkillebrew-g, 2020-12-12:
So that we can do things like:

```
const NUM_BLAH = u32:3; 
type Blah = uN[clog2(NUM_BLAH)];
```

and `Blah` will be 2 bits wide.

cdleary, 2021-01-05:
Believe Rob is on the case for this one, but not sure in particular for its use in a typedef as an expression, you may need to bind the expression to an identifier and then use it in the typedef.

dkillebrew-g, 2021-01-06:
> not sure in particular for its use in a typedef as an expression

In the long term, hopefully this is supported too :)  Your suggested workaround will be fine until then, I'm sure.

cdleary, 2021-01-08:
> > not sure in particular for its use in a typedef as an expression
> 
> In the long term, hopefully this is supported too :) Your suggested workaround will be fine until then, I'm sure.

Although it's a somewhat arbitrary restriction, it seems kind of ugly to be able to have type annotations with arbitrary expression in them everywhere, I'm curious if there's some aspect of "that'll be super useful" that you're thinking of that I'm not thinking of, or if it's more of a "if it's reasonably possible might as well give me the expressiveness" kind of lines of reasoning? Thanks for explaining Dan!

dkillebrew-g, 2021-01-08:
> it seems kind of ugly to be able to have type annotations with arbitrary expression in them everywhere

IMO being able to name arbitrary expressions, when they're used to define a type, improves readability:

- It's good practice to name expressions that may be confusing to the reader; the writer can name it and provide clarity to the reader. The code becomes self documenting. The name is used everywhere (whereas a comment that explains a complex expression would likely only be written once, not everywhere the complex expression is used). Let's reduce the mental burden on the reader. Read the expression and its name once, and understand what the name means. Reuse that understanding next time the name is seen... instead of having to process the complex expression for the Nth time.

- When the writer names an expression that is repeated, this alleviates the reader from asking the question, "have I seen this expression before? is it the same expression that I saw used elsewhere? Or is it subtly different?" I.e. the reader no longer must perform pattern recognition and deduplication. Let's reduce the mental burden on the reader.

Being able to name an arbitrary expression improves readability. Why should it matter if that expression is being used to define a type? We allow naming of arbitrary expressions that define values... why should types be treated differently here?

Just drawing from my C++ and Haskell experience, it is not rare for me to encounter a type that is used more than once. It's not rare for me to encounter type-expressions that are complex, where naming the type improves readability.

To your question, I firmly think this is in the former category. This is useful. I've used it, and I've benefited from others using it in real world code, many times over.

cdleary, 2021-01-08:
I think maybe we're not talking about the same thing, naming expressions means "creating a binding" right? That is:

```
const NUM_BLAH = u32:3; 
type Blah = uN[clog2(NUM_BLAH)];
```

Implies you can use a type `uN[clog2(NUM_BLAH)]` in all places in the grammar that accept a type expression. My question, which I should have made clearer, was, does it seem like a harmful restriction to only permit:

```
const NUM_BLAH = u32:3;
const NUM_BLAH_BITS = clog2(NUM_BLAH); 
type Blah = uN[NUM_BLAH_BITS];
```

i.e. do we want type expressions to always be able to take arbitrary expressions, or is it reasonable to encourage (or perhaps /only/ permit) expressions used in type annotations to be defined (via binding) outside of the type expression syntax itself.

Not objecting to binding things, which I think is the position you're responding to with your comment -- the opposite, I'm actually saying "is it so bad if we force people to bind things outside of the type expressions to make sure type expression syntax is simpler". Or, put another way, if we had the limitation that you /had/ to bind outside the type expression syntax and just make references to names, how bad of a limitation is that for expressiveness.

Doesn't matter too much at the end of the day, just wanted to clarify.

dkillebrew-g, 2021-01-08:
> does it seem like a harmful restriction to only permit

I don't think this is a helpful restriction. So yes, (somewhat) harmful. 

Naming is hard. The restriction you propose requires the writer to name the intermediate expression, here named NUM_BLAH_BITS. 

```
const NUM_BLAH_BITS = clog2(NUM_BLAH); 
type Blah = uN[NUM_BLAH_BITS];
```

Perhaps the name to use here is obvious, but that's far from true in every case when programming. I think it's better to give the writer the choice of where and when to name things. Don't force them to come up with names when they don't want to or can't.

----

The example we're using here is relatively simple and straightforward, but with full support for parametric types, N-dimensional arrays and what have you, types will not be this simple.

----

> make sure type expression syntax is simpler

I'm not convinced this is simpler. 

If we allow arbitrary type-expressions on the right hand side of the `type NAME =` statement, then isn't it simple to say "the  type-expression on the right hand side follows the exact same syntax & semantics as a type expression that you'd use, for example, in a function argument and/or result type"?

Whereas I think it's more complex (because there's less consistency and uniformity), if we instead say, "the right hand side of a type alias must be a primitive type or user-defined type, and array sizes, or parametric arguments must be constants that are defined elsewhere. Type expressions used in function arguments and results don't have these restrictions".

cdleary, 2021-01-08:
They should be the same (I was not saying they would have two different sets of restrictions, all type expressions would have the same restriction) -- the only thing I was raising was that the DSL may require you to have a binding the value-based components of the type expression to avoid the value expression part of the grammar being intermixed arbitrarily in the type expression part of the grammar. It would probably be a temporary restriction in any case, but I was actually mulling that it has some nice readability properties. It's clear you disagree, but we may have it for a while and I was looking for feedback on prioritization. Thanks for the feedback!

dkillebrew-g, 2021-01-08:
> all type expressions would have the same restriction

Oh. I think I see now what you're saying.

> the DSL may require you to have a binding for the value-based components of the type expression to avoid the value expression part of the grammar being intermixed arbitrarily in the type expression part of the grammar. It would probably be a temporary restriction in any case

Temporary restrictions are fine, especially for things like this, that have reasonable/easy workarounds. The project is still new and rapidly implementing all kinds of new functionality. :)

---

In terms of prioritization, I agree with your interpretation that there exist a number of higher priority things, at this moment.

cdleary, 2021-01-09:
Excellent, thanks Dan!

closed 2021-03-11 by copybara-service[bot]