google/xls #234: XLS pulls in two different re2 libraries

meheff, 2020-12-21:
We directly use @com_google_re2, but grpc appears to pull in @com_github_google_re2. If a build target depends on both... boom!

Perhaps we should change to using @com_github_google_re2 even though @com_google_re2 is more canonical.

hongted, 2020-12-21:
Updated to @com_github_google_re2 with 70a1627

closed 2020-12-21 by hongted

per-gron, 2021-04-09:
Note: `http_archive` (and related rules) have a `repo_mapping` attribute that can be used for adding workspace aliases for imported workspaces. This can help work around issues like this when there are conflicts.

cdleary, 2022-01-22:
Just a note that I couldn't get repo_mapping to work for grpc since it used `bind` to address the name of the `http_archive`:

https://github.com/grpc/grpc/blob/b4b6862352f929d1d1543c1fa3d16ff3524bf459/bazel/grpc_deps.bzl#L123

```starlark
    native.bind(
        name = "re2",
        actual = "@com_googlesource_code_re2//:re2",
    )
```

Additionally it tested whether the native rule with exact name being present in `native.existing_rules()` -- I think these are "anti-patterns" in the sense they prevent `repo_mapping` from working, but I'm not really 100% sure, but I know the patch approach seemed to work ok in https://github.com/google/xls/commit/8a0fbbd3066a7bf9a5ddfef7d6e7ed8a5b4f3b57#diff-b84a00fb8067122ea8c4f5c2b704bbb4ed579cd1a95eb43206c9c5cc2fd14d67

```starlark
    if "com_googlesource_code_re2" not in native.existing_rules():
        http_archive(
            name = "com_googlesource_code_re2",
            sha256 = "319a58a58d8af295db97dfeecc4e250179c5966beaa2d842a82f0a013b6a239b",
            # Release 2021-09-01
            strip_prefix = "re2-8e08f47b11b413302749c0d8b17a1c94777495d5",
            urls = [
                "https://storage.googleapis.com/grpc-bazel-mirror/github.com/google/re2/archive/8e08f47b11b413302749c0d8b17a1c94777495d5.tar.gz",
                "https://github.com/google/re2/archive/8e08f47b11b413302749c0d8b17a1c94777495d5.tar.gz",
            ],
        )
```

cdleary, 2022-01-22:
Just to clarify that last entry with a TL;DR with less details: got re2 _working_ after the upgrade, but had to patch up grpc as part of the build flow to do it.