google/xls #242: [XLScc] C++ frontend cannot handle array declarations with operands in declaration

ted-xie, 2020-12-29:
githash: `e84a307f7f9ce064748b966c5a3b3525abb0cd03`

Repro example:

```
$ cat define_test.cc 
void mytest(char vec[128+128]) {}

$ bazel-bin/xls/contrib/xlscc/driver/xlscc define_test.cc  define_test.ir
Traceback (most recent call last):
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/driver/main.py", line 94, in <module>
    main()
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/driver/main.py", line 88, in main
    translator.parse(ast)
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/translate/translator.py", line 518, in parse
    func.parse_function(self, child)
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/translate/translator.py", line 358, in parse_function
    t = translator.parse_type(child.type)
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/translate/translator.py", line 605, in parse_type
    size, size_type = parse_constant(ast.dim)
  File "/home/tedx/sw/xls/bazel-bin/xls/contrib/xlscc/driver/xlscc.runfiles/com_google_xls/xls/contrib/xlscc/translate/translator.py", line 423, in parse_constant
    if stmt_ast.type == "int" or stmt_ast.type == "long":
AttributeError: 'BinaryOp' object has no attribute 'type'

```
The preprocessor should be robust enough to handle some compile-time known constants in the declaration.

spurserh, 2020-12-29:
Hey Ted, 

I'm just moving the new clang-based XLS[cc] into github now, which will solve this problem. 

I'll update the bug when it's ready. Should be in the next week. 

- Sean

spurserh, 2021-03-26:
Please retry with new XLS[cc] https://github.com/google/xls/blob/main/xls/contrib/xlscc/README.md

cdleary, 2021-09-21:
Closing all the bugs filed against the old xlscc, if this same issue is encountered with the Clang-based FE we'll file a new bug / make a new test case.

closed 2021-09-21 by cdleary