google/xls #243: [booleanify] Unsupported operation sgt

spurserh, 2021-01-05:


cdleary, 2021-01-05:
Sean do you have a sample input file / command on hand from when you observed the issue that would make it easier to repro? Assuming this one goes to @RobSpringer

spurserh, 2021-01-05:
package my_package

fn foo(a: bits[24]) -> (bits[24], bits[8]) {
  literal.1557: bits[24] = literal(value=1, id=1557, pos=2,71,5)
  aid8id12__1: bits[24] = add(a, literal.1557, id=1526, pos=2,598,3)
  literal.1599: bits[1] = literal(value=0, id=1599, pos=2,300,5)
  offsetp_aid2: bits[4] = bit_slice(aid8id12__1, start=20, width=4, id=1556, pos=2,698,5)
  concat.1598: bits[5] = concat(literal.1599, offsetp_aid2, id=1598, pos=2,300,5)
  literal.1589: bits[5] = literal(value=0, id=1589, pos=2,300,5)
  literal.1736: bits[1] = literal(value=0, id=1736, pos=2,300,5)
  offsetp_aid2id24__1p_aid22: bits[1] = sgt(concat.1598, literal.1589, id=1567, pos=2,300,5)
  literal.1590: bits[5] = literal(value=1, id=1590, pos=2,300,5)
  concat.1706: bits[2] = concat(literal.1736, offsetp_aid2id24__1p_aid22, id=1706)
  literal.1720: bits[2] = literal(value=2, id=1720, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__1: bits[1] = sgt(concat.1598, literal.1590, id=1570, pos=2,300,5)
  opid28id30__15: bits[2] = or(concat.1706, literal.1720, id=1679, pos=2,605,3)
  literal.1737: bits[1] = literal(value=0, id=1737, pos=2,300,5)
  sel.1643: bits[2] = sel(offsetp_aid2id24__1p_aid22__1, cases=[concat.1706, opid28id30__15], id=1643)
  literal.1591: bits[5] = literal(value=2, id=1591, pos=2,300,5)
  concat.1708: bits[3] = concat(literal.1737, sel.1643, id=1708)
  literal.1723: bits[3] = literal(value=4, id=1723, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__2: bits[1] = sgt(concat.1598, literal.1591, id=1573, pos=2,300,5)
  opid28id30__16: bits[3] = or(concat.1708, literal.1723, id=1684, pos=2,605,3)
  literal.1738: bits[1] = literal(value=0, id=1738, pos=2,300,5)
  sel.1649: bits[3] = sel(offsetp_aid2id24__1p_aid22__2, cases=[concat.1708, opid28id30__16], id=1649)
  literal.1592: bits[5] = literal(value=3, id=1592, pos=2,300,5)
  concat.1710: bits[4] = concat(literal.1738, sel.1649, id=1710)
  literal.1726: bits[4] = literal(value=8, id=1726, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__3: bits[1] = sgt(concat.1598, literal.1592, id=1576, pos=2,300,5)
  opid28id30__17: bits[4] = or(concat.1710, literal.1726, id=1689, pos=2,605,3)
  literal.1739: bits[1] = literal(value=0, id=1739, pos=2,300,5)
  sel.1655: bits[4] = sel(offsetp_aid2id24__1p_aid22__3, cases=[concat.1710, opid28id30__17], id=1655)
  literal.1593: bits[5] = literal(value=4, id=1593, pos=2,300,5)
  concat.1712: bits[5] = concat(literal.1739, sel.1655, id=1712)
  literal.1729: bits[5] = literal(value=16, id=1729, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__4: bits[1] = sgt(concat.1598, literal.1593, id=1579, pos=2,300,5)
  opid28id30__18: bits[5] = or(concat.1712, literal.1729, id=1694, pos=2,605,3)
  literal.1740: bits[1] = literal(value=0, id=1740, pos=2,300,5)
  sel.1661: bits[5] = sel(offsetp_aid2id24__1p_aid22__4, cases=[concat.1712, opid28id30__18], id=1661)
  literal.1594: bits[5] = literal(value=5, id=1594, pos=2,300,5)
  concat.1714: bits[6] = concat(literal.1740, sel.1661, id=1714)
  literal.1732: bits[6] = literal(value=32, id=1732, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__5: bits[1] = sgt(concat.1598, literal.1594, id=1582, pos=2,300,5)
  opid28id30__19: bits[6] = or(concat.1714, literal.1732, id=1699, pos=2,605,3)
  literal.1741: bits[1] = literal(value=0, id=1741, pos=2,300,5)
  sel.1667: bits[6] = sel(offsetp_aid2id24__1p_aid22__5, cases=[concat.1714, opid28id30__19], id=1667)
  literal.1595: bits[5] = literal(value=6, id=1595, pos=2,300,5)
  concat.1716: bits[7] = concat(literal.1741, sel.1667, id=1716)
  literal.1735: bits[7] = literal(value=64, id=1735, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__6: bits[1] = sgt(concat.1598, literal.1595, id=1585, pos=2,300,5)
  opid28id30__20: bits[7] = or(concat.1716, literal.1735, id=1704, pos=2,605,3)
  literal.1742: bits[1] = literal(value=0, id=1742, pos=2,300,5)
  sel.1673: bits[7] = sel(offsetp_aid2id24__1p_aid22__6, cases=[concat.1716, opid28id30__20], id=1673)
  literal.1596: bits[5] = literal(value=7, id=1596, pos=2,300,5)
  concat.1717: bits[8] = concat(literal.1742, sel.1673, id=1717)
  literal.1564: bits[8] = literal(value=128, id=1564, pos=1,28,4)
  offsetp_aid2id24__1p_aid22__7: bits[1] = sgt(concat.1598, literal.1596, id=1588, pos=2,300,5)
  opid28id30__14: bits[8] = or(concat.1717, literal.1564, id=1547, pos=2,605,3)
  sel.1163: bits[8] = sel(offsetp_aid2id24__1p_aid22__7, cases=[concat.1717, opid28id30__14], id=1163)
  ret tuple.1501: (bits[24], bits[8]) = tuple(aid8id12__1, sel.1163, id=1501, pos=1,10,8)
}

RobSpringer, 2022-04-25:
Unsure when this was resolved, but the provided IR now booleanifies successfully.

closed 2022-04-25 by RobSpringer