google/xls #246: DSLX: Move constexpr evaluation to typechecking

RobSpringer, 2021-01-06:
Shortly, I'll submit a change that enables "constexpr" function & constant evaluation in the DSL. That initial drop will have the evaluation code in the IR converter, but from internal discussions, it makes more sense for it to live in the typechecker.

In the interest of landing this long-in-coming change, we'll submit as-is, but then work to move it into the typechecker. Not sure what the precedence will be, given the migration to C++.

cdleary, 2021-01-06:
Hmm, point of clarification, in the current state can you use all top level consts in type aliases? e.g.

```
const N = u32:42;
const NP1 = N + u32:1;
type MyBits = bits[NP1];
```

I thought the answer was yes (just that the IR converter and the type checking will redundantly call the interpreter to determine const values, instead of noting what expressions are const and what their values are one at typechecking time) but I want to be sure I got that right. Thanks!

cdleary, 2021-01-21:
Reasonable to think this is a blocking issue for #207 and #259 in that we should resolve it first.

cdleary, 2021-01-30:
https://github.com/google/xls/commit/1ff7d752e41a22f8b386b7fbea481de5b8d3f901 Was towards this as well (more places in the grammar where we'll accept exprs but they must be evaluated as compile time constants), forgot to tag the issue.

cdleary, 2021-03-02:
Example of dimension that originates from an another-module constant reference: https://github.com/google/xls/blob/b0dba24192e16add1ffa51bd2128f7ab2f8ba0cf/xls/dslx/tests/mod_const_in_type_alias.x
Example of it being referred to directly: https://github.com/google/xls/blob/4ed83da9f521e8e79085d15070d1eedaf07454ff/xls/dslx/tests/mod_const_in_type_alias.x
Example of doing some constexpr math on another-module constant reference and using that as a dimension in a type alias: https://github.com/google/xls/blob/b0dba24192e16add1ffa51bd2128f7ab2f8ba0cf/xls/dslx/tests/mod_const_in_type_alias_expr.x#L19

closed 2021-03-02 by cdleary