google/xls #247: DSLX: Explicit struct params not validated

RobSpringer, 2021-01-06:
Consider the following code sequence:
```
struct ParametricPoint<A: u32, B: u32 = double(A)> {
  x: bits[A],
  y: bits[B]
}

fn foo() {
  let bar =
      ParametricPoint<bits[666]:666, bits[777]:777>{ x: u8:15, y: u16:18 };
  ()
}
```

I don't believe this should compile, as the explicit parametrics are of bits[666] and bits[777], which do NOT match the types given in the struct definition. Currently, in the parser, explicit struct parametrics are parsed for syntactic validity but aren't checked against the StructDef's parametric bindings.

cdleary, 2021-06-23:
Rob said I could steal this one. :-)

cdleary, 2021-06-24:
Example of something very simple that should flag an error -- even though there's a "phantom" parametric (unused in the body of the struct definition), they are _structurally_ the same but not _nominally_.

```
struct MyStruct<N: u32> {}
fn main(x: MyStruct<u32:8>) -> MyStruct<u32:42> { x } 
```

Think I'm going to prefactor `ConcreteType` first, right now there's a "nominal" field for the struct definition hanging off `TupleType` (with member names) but worth breaking out into its own subclass first, since we now want to bolt parametrics into it for nominal comparisons.

richmckeever, 2024-07-26:
I think only the case with the phantom parametric is broken now.

The original example, with `bits[666]:666` as a parametric, is going to error with `must be a `u32` (soon to be `usize`...`, and there will be a different error about inconsistent args if an instantiation contradicts the `B:u32 = double(A)`.

closed 2024-09-03 by richmckeever