google/xls #248: DSLX: Module::ToString() generates hard-to-read output

RobSpringer, 2021-01-06:
For an example, see xls/tools/testdata/proto_to_dslx_main.x, which was generated from such a call.

```
pub const Foo = Fields { fields: [Field { index: bits[32]:0, bit_offset: bits[32]:0, width: bits[32]:4, foo: [bits[64]:1, bits[64]:2, bits[64]:3, bits[64]:4], foo_count: u32:4, sub_fields: [SubField { sub_index: bits[32]:1 }, SubField { sub_index: bits[32]:2 }, SubField { sub_index: bits[32]:3 }, SubField { sub_index: bits[32]:4 }], sub_fields_count: u32:4 }, Field { index: bits[32]:1, bit_offset: bits[32]:4, width: bits[32]:5, foo: [bits[64]:1, bits[64]:2, bits[64]:3, bits[64]:0], foo_count: u32:3, sub_fields: [SubField { sub_index: bits[32]:1 }, SubField { sub_index: bits[32]:2 }, ...
```

It would be much nicer if we had some line breaks and indentation in all that.

cdleary, 2021-01-06:
This is probably a special case of https://github.com/google/xls/issues/25 so I'm not sure it's _necessarily_ a good first issue... or maybe it is, if it makes a reusable bit of functionality that could go to the broader formatter -- in general making a mechanism that introduces newlines and appropriate indentation for expressions that are longer than nominal colwidth (generally 80 chars in Goog style guide) would be the nice-to-have here and we could just filter through that dslx-format helper.

cdleary, 2021-02-19:
I'm gonna declare this duplicate of #25 since we want something that will generally auto-format DSL constructs.

closed 2021-02-19 by cdleary