google/xls #256: Add capability to reuse expensive resources

meheff, 2021-01-12:
This would be a transformation which combines identical/similar operations by muxing in different operands. This can only be done if only one of the operations is being executed either because only one is live at a time, or somehow time multiplexed. The transformation likely needs to be schedule aware to avoid unacceptably increasing critical path length. This is a pretty deep area with lots of space to explore.

vincent-mirian-google, 2021-01-12:
Yes. Resource sharing is an interesting space. Also, we can explore multi-pumping the inputs to the resource. "Multi-pumping is a method whereby a resource is clocked at a frequency that is a multiple of the surrounding circuit, thereby offering multiple executions per global clock, and therefore sharing [the resource] in the same clock cycle."

cdleary, 2021-01-12:
@vincent-mirian-google Sounds interesting, maybe we could file a separate bug for that -- this would require a new clock domain presumably? But so long as its integral multiples some reasoning gets easier that arbitrary clock domain crossing I assume? (Note everything we've been doing so far has been single clock domain, though with procs we may be able to have a nice clean interface to put clock-crossing elements on.)

This bug I think is primarily targeting things like opcodes or sub-opcodes, where disjointly you either use path X or path Y, and then consolidating the resources required for those paths (e.g. in the scheduler's resource mapper) because you know X and Y are disjoint. In an FSM you can consider all the states disjoint, but for a pipeline at least there's a tradeoff in the pareto space (latency v area) between whether you want to change the schedule to share disjoint resources more, potentially at the cost of latency. Gets into "what should we explore if we offer a latency/area knob (or set of knobs)". @meheff did I capture that right?

meheff, 2021-01-13:
@cdleary That's basically right. Though I think the trade off in a pipeline for sharing resources is more throughput vs area than latency vs area. Reusing resources reduces throughput vs a fully pipelined design.

cdleary, 2021-01-13:
Are you thinking that's also the case in a fully pipelined design without time multiplexing (II=1)? I was thinking "well maybe I have opportunities to reuse more but [e.g. by causing these two resources used by disjoint opcodes to both be in C1 instead of one being in C0 and one in C1] it will elongate the critical path". Definitely more reuse via time multiplexing or increasing initiation interval would reduce throughput though, just checking if I'm making an error in my mental model for your thinking on thoughput vs area being the primary tradeoff space.

meheff, 2022-01-06:
This could be (should be?) driven by user hints indicating the desire to share the resource. One way of doing this might be to have a function-level annotation that says "only stamp out one of me". If the function is inlined in multiple places then XLS would guarantee that the nodes in the function (like a multiply node) would not be cloned. Instead operands would be muxed in from different locations. If for some reason, this could not be done, then XLS would raise an error.

An alternative is to annotate individual operations, but the advantage of a function is that it provides a nice source-level construct to attach things to. Where would you put an annotation on the '*' operator?

taktoa, 2022-01-10:
Another thought: at II=1, could we still get this for power gated parts of the circuit? If we know that the gating of two parts of the graph is mutually exclusive, we can share resources between them for a big area win (assuming of course that people actually use gating that way)

vincent-mirian-google, 2022-01-11:
@taktoa Good point.

@meheff Quick question (maybe something to consider?): Background: there is no 'inline' in dslx, so I assume you mean an invocation. Regarding procs: When a function with a resource constraint is called by proc and there are multiple procs, I assume that there will be one instantiation of the resource per proc?

allight, 2026-11-14:
Keeping this issue open (reviewed by @allight): Although `ResourceSharingPass` in `xls/passes/resource_sharing_pass.cc` (with `xls/passes/folding_graph.cc`, `xls/passes/resource_sharing_equivalence.cc`, and `xls/scheduling/mutual_exclusion_pass.cc`) implements initial resource sharing, we are not closing this issue because we want to continue improving resource sharing. (Summarized by Gemini)