google/xls #259: Imported constants are not usable in type expressions

dkillebrew-g, 2021-01-15:
I'd like to do this:

```
fn do_foo(x : MyType) -> uN[types::SIZE] {
```

where in some other file, e.g. `types.x` we've defined a public constant named `SIZE`.

but that causes a parse error
```
ParseError: importer.x:88:54-88:56 Expected ']', got '::'
```

and so I'm forced to do this workaround:

```
const SIZE = types::SIZE;
fn do_foo(x : MyType) -> uN[SIZE] {
```

dkillebrew-g, 2021-01-15:
Actually, even the workaround doesn't work, so editing the above...

dkillebrew-g, 2021-01-15:
Workaround: export the entire type (as opposed to the constant that would be used in the type's declaration). E.g. export this:
```
pub type SizeThing = uN[SIZE];
```

cdleary, 2021-01-15:
Tentatively assigning Rob -- good example that shows how hoisting constexpr evaluation to the typechecker will enable use in both in the typesystem and the IR converter (right now the things that were added for constexpr evaluation were confined to the IR converter), but if Rob is swamped I may be able to grab it too, LMK!

closed 2021-03-10 by copybara-service[bot]